embedded-graphics/embedded-graphics #469: Remove variable width glyph support for internal font renderer?

rfuest, 2020-11-10:
Do we still need the variable width support for the internal renderer if we support more advanced external renderers? To my knowledge only one font that uses this feature exits: `Font6x6`.

I wanted to rename the current `Font` trait to `MonospacedFont`, but that name wouldn't make sense if we keep the variable width support. Any suggestions for a better one? The docs also refer to the fonts as "monospaced fonts" which isn't accurate anymore.

jamwaffles, 2020-11-10:
Sounds good to me. I think e-g should provide "I just need to print some stuff"-level text support, so variable width fonts would make more sense to live outside e-g. There's also #268 which could be closed.

rfuest, 2020-11-10:
Great, I'll remove the variable width support and `Font6x6`. I think #443 can also be closed, because in a monospaced font a space is expected to be the same width as every other character.

jamwaffles, 2020-11-10:
I think so too.

It would be good to find a way of providing similar features elsewhere, but I think that might be something for `embedded-text` to explore (or the BDF fonts, or another text library, etc)

rfuest, 2020-11-10:
> It would be good to find a way of providing similar features elsewhere, but I think that might be something for `embedded-text` to explore (or the BDF fonts, or another text library, etc)

I hope that the font renderer API will be flexible enough to make this possible. The BDF renderer will support all features required to draw a font similar to `Font6x6` with a larger space width.

Another thing I'm considering to remove is the support for multiline text. If we can integrate the font renderer API into e-t we have a much more flexible solution. And the current implementation would require all text renderers to support our basic multiline text, which adds a considerable complexity to all text rendering crates. Thoughts?

But I'm also considering to add a feature we had considered before, but decided not to implement: `FrameText`. `FrameText` would render text inside a box with an optionally colored background. This might sound like it conflicts with what I said in the previous paragraph, but the internal `FrameText` would be much simpler than what e-t implements and, for example, only support a single line of text. And I think that e-t could benefit from `FrameText`, because it can reuse the same API to the font renderers.

The digital time display in `analog-clock.rs` is an example https://github.com/embedded-graphics/simulator/blob/da56ee43c37765c30ac084b1aac91bae782cfb1a/examples/analog-clock.rs#L129 of why this would be useful. Setting the `background_color` on a `Text` object doesn't result in a good looking output for most fonts. For this reason the example draws a rectangle (`fill_solid`) before the text is drawn on top of it (`draw_iter`). If we would implement a `FrameText` object the text with background could be drawn using a single `fill_contiguous` draw call.

I haven't figured out the exact feature set a `FrameText` implementation would have, but I would like some feedback if this is an idea that is worth pursuing.

jamwaffles, 2020-11-10:
> Another thing I'm considering to remove is the support for multiline text. If we can integrate the font renderer API into e-t we have a much more flexible solution. And the current implementation [...]

I would like to keep support for multiline text, but you're right that it would then become a requirement for every renderer. I can live without it, although it's very helpful for "just print these vars" debugging.

`FrameText` would be a good feature. I mentioned anchor points long ago in #225. I assume you wouldn't implement these as part of the new feature? Nothing else in e-g has anchor points so makes sense to continue that pattern. Other crates can handle layout I suppose :)

Would normal styled text lose the ability to have a background colour set?

> If we would implement a FrameText object the text with background could be drawn using a single fill_contiguous draw call.

Very nice!

> I haven't figured out the exact feature set a FrameText implementation would have, but I would like some feedback if this is an idea that is worth pursuing.

It might be a good idea to spec out the list of features you have in mind as a starting point. I think my main question right now is what happens to the current styled text impl. Do you see `FrameText` being a "building block" component and quite simple?

rfuest, 2020-11-10:
> I would like to keep support for multiline text, but you're right that it would then become a requirement for every renderer. I can live without it, although it's very helpful for "just print these vars" debugging.

Perhaps there is an alternative implementation that doesn't require each render to implement multiline text. I'll look into this.

> `FrameText` would be a good feature. I mentioned anchor points long ago in #225. I assume you wouldn't implement these as part of the new feature? Nothing else in e-g has anchor points so makes sense to continue that pattern. Other crates can handle layout I suppose :)

I haven't decided yet, some settings for alignment and padding would make the feature more useful, but also more complex. I will focus on the rendering side first and think about the user API later.

> Would normal styled text lose the ability to have a background colour set?

Would you mind if I did remove this feature? I was considering to remove it, but perhaps it's still useful for a quick debug output or fonts that are designed to be used this way (e.g. by having a consistent padding around the chars).

> It might be a good idea to spec out the list of features you have in mind as a starting point. I think my main question right now is what happens to the current styled text impl.

I'll open a draft PR with the changes to the styled text impl sometime this week. From a user perspective the only change is that `TextStyle` and `TextStyleBuilder` were renamed to `MonospacedTextStyle` and `MonospacedTextStyleBuilder`. Each text renderer will use its own text style and is able to add any necessary styling attributes this way.

> Do you see `FrameText` being a "building block" component and quite simple?

I'm not sure what you mean by "building block" in this context. I'll wait with the final design of `FrameText` until I'm finished with the refactor of the normal `Text` rendering. They should be relatively independent of each other.

rfuest, 2020-11-10:
> Nothing else in e-g has anchor points so makes sense to continue that pattern.

I was thinking about adding a `resize` function to `Rectangle` and maybe the other primitives, that would change the size relative to an anchor point. So maybe it would be a good idea to add anchors.

bugadani, 2020-11-13:
I just want to mention (though I'm certain you're aware of it), that the current variable width text rendering is rather dumb. It will always render spaces with the same width as character spacing, which makes text hardly legible as there isn't enough space between words. I'm all for getting rid of it in it's current form, and it's something I wouldn't mind removing from e-t as well.

jamwaffles, 2020-11-16:
I think this is closed by #470?

rfuest, 2020-11-16:
The original issue was resolved by #470. The discussion also contains some talk about `FrameText`, but I can open a new issue for that feature.

closed 2020-11-16 by rfuest