embedded-graphics/embedded-graphics #471: Polyline end segment out of bounds draw

jamwaffles, 2020-11-13:
- Version of embedded-graphics in use (if applicable): [version here]
- Target: simulator

## Description of the problem/feature request/other

For _some_ positions and thicknesses of polyline points, the end segment has screwed up rendering:

[image]

Only happens at `stroke: 5` as far as I can tell. Ignore the triangle and horizontal blue line - that's from other test stuff.

Do bear in mind this is on a WIP branch where I've been messing around with line joints, so may not be reproducible in master - I'll check again when I'm in a clean enough state to `git checkout master`.

## Test case (if applicable)

```rust
let line_style = PrimitiveStyle::with_stroke(Rgb888::GREEN, 5);

let h = 100;

let mouse_pos = Point::new(137, 15);

let points = [
    Point::new(10, h / 2),
    Point::new(50, h / 2),
    Point::new(60, h / 2 - 20),
    Point::new(70, h / 2),
    Point::new(80, h / 2),
    Point::new(90, h / 2 + 10),
    Point::new(100, 10),
    Point::new(110, h / 2 + 20),
    Point::new(120, h / 2),
    mouse_pos,
];

Polyline::new(&points)
    .into_styled(line_style)
    .draw(display)?;
```

rfuest, 2020-11-14:
You might have already checked, but I've just tested it in master and the error also exists there. There seems to be a problem with odd stoke widths when three points are almost on a straight line.

[image]

```rust
for width in 1..20 {
    let pos = Point::new(width as i32 * 30, 0);

    Polyline::new(&points)
        .into_styled(PrimitiveStyle::with_stroke(Rgb888::GREEN, width))
        .draw(&mut display.translated(pos))?;

    Text::new(&width.to_string(), pos + Point::new(10, 80))
        .into_styled(TextStyle::new(Font6x8, Rgb888::WHITE))
        .draw(&mut display)?;
}

// 3 points on a straight line -> works
let points2 = [Point::new(10, 70), Point::new(20, 50), Point::new(30, 30)];

for width in 1..20 {
    let pos = Point::new(width as i32 * 30, 100);

    Polyline::new(&points2)
        .into_styled(PrimitiveStyle::with_stroke(Rgb888::RED, width))
        .draw(&mut display.translated(pos))?;

    Text::new(&width.to_string(), pos + Point::new(10, 80))
        .into_styled(TextStyle::new(Font6x8, Rgb888::WHITE))
        .draw(&mut display)?;
}
```

jamwaffles, 2020-11-14:
Completely forgot to check in master :grimacing: thanks for the extra test case!

jamwaffles, 2020-11-14:
Looks like it's in the `LineJoin` code somewhere. The middle join in the line is producing a `first_edge_end.left` and (they're the same) `second_edge_start.left` point of `32,5` which kicks it past the end. Will do a bit more digging.

jamwaffles, 2020-11-14:
Further narrowed down to https://github.com/embedded-graphics/embedded-graphics/blob/master/src/primitives/line_join.rs#L158

jamwaffles, 2020-11-21:
This has mostly mysteriously fixed itself, although now exhibits slightly different behaviour. Notice the right-side cap on the 19px wide green line:

[image]

The widths for the second segment are still a pixel different, but I'm not sure we can do much about that without resorting to floating point.

jamwaffles, 2020-11-21:
Changing the non working points to `let points = [Point::new(10, 70), Point::new(20, 50), Point::new(31, 30)];` now gets this result:

[image]

rfuest, 2020-11-21:
> This has mostly mysteriously fixed itself,...

`git bisect` revealed that the mysterious fix was included in this commit: c04430b9ae62cb773432535cc0b3fcb382ceab50

jamwaffles, 2020-11-30:
With the `LinearEquation`s, is there a way of testing at which end of the line segment the intersection occurs? The issues above are caused by the intersection point being found on the "wrong end" of the line segments. One solution I thought up but failed to implement so far is to add a check to see where the intersections fall on both lines `l1` and `l2`, where `l1` is the right edge of the first polyline segment, and `l2` is the right edge of the second segment.

We could see if the intersection is on `l1` or _after_ `l1.end`, **and** on `l2` or _before_ `l2.start`. This would guarantee the intersection is constrained to both lines and the space between them.

If the intersection is on the wrong end of the lines as defined above, we could probably assume the lines are almost parallel at this point, and just pick the average between `l1.end` and `l2.start`.

rfuest, 2020-12-01:
I don't think there is an easy way to tell on which side the intersection occurred using our current method.

The link you've added in the `intersection` method shows a method to implement this: http://paulbourke.net/geometry/pointlineplane/#i2l. The condition described in the last note could be extended to also include intersections after `l1.end` and `l2.start` by checking `u1 >= 0 && u2 <= 1`. But it would require some scaling or floating point to get the required resollution.

commit 17a231c5 (2021-01-22): Detect nearly-parallel numerical inaccuracy

Not too happy with this solution, but it fixes the original problem so
at least there's that.

I don't understand the maths behind it at _all_, this just seemed to
work for the cases in #471. Hopefully no other bugs are introduced by
this change.

I'm very open to a better solution, or at least an explanation of why my
fix works.

Closes #471.
commit fe97aeb5 (2021-01-22): Detect nearly-parallel numerical inaccuracy

Not too happy with this solution, but it fixes the original problem so
at least there's that.

I don't understand the maths behind it at _all_, this just seemed to
work for the cases in #471. Hopefully no other bugs are introduced by
this change.

I'm very open to a better solution, or at least an explanation of why my
fix works.

Closes #471.
commit 2d452b06 (2021-01-25): Detect nearly-parallel numerical inaccuracy

Not too happy with this solution, but it fixes the original problem so
at least there's that.

I don't understand the maths behind it at _all_, this just seemed to
work for the cases in #471. Hopefully no other bugs are introduced by
this change.

I'm very open to a better solution, or at least an explanation of why my
fix works.

Closes #471.
PR #527 (2021-01-25): Detect nearly-parallel numerical inaccuracy
Thank you for helping out with embedded-graphics development! Please:

- [x] Check that you've added passing tests and documentation
- [x] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, etc) if your changes affect the **public API**
- [x] Run `rustfmt` on the project
- [x] Run `just build` (Linux/macOS only) and make sure it passes. If you use Windows, check that CI passes once you've opened the PR.

## PR description

Not too happy with this solution, but it fixes the original problem so [... cut]
closed 2021-01-25 by jamwaffles