embedded-graphics/embedded-graphics #473: Thick polylines are drawn without translation

rfuest, 2020-11-14:
Polylines with a stroke width > 1px don't use the `Polyline::translate` field and are always drawn without an offset.

Should we fix this issue or remove the `translate` field entirely, because polylines can also be translated by using a `TranslatedDrawTarget`? This would be a cleaner implementation IMO, but we would loose the ability to get the bounding box of a translated styled polyline.

jamwaffles, 2020-11-14:
I'll look into this more later, but my immediate reaction is to keep the translatability of polylines to keep the surprise factor low - this way they keep similar features to other primitives.

rfuest, 2020-11-14:
OK, but we could still use a translated draw target internally to reduce code duplication and remove the performance overhead for non translated polylines. The `Drawable` impl for styled polylines could be extended to directly draw to given draw target for non translated polylines or a translated draw target if the polyline is translated.

PR #495 (2020-11-21): Fix drawing of translated polylines
Thank you for helping out with embedded-graphics development! Please:

- [x] Check that you've added passing tests and documentation
- [x] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, etc) if your changes affect the **public API**
- [x] Run `rustfmt` on the project
- [x] Run `just build` (Linux/macOS only) and make sure it passes. If you use Windows, check that CI passes once you've opened the PR.

## PR description

Closes #473
closed 2020-11-21 by jamwaffles