embedded-graphics/embedded-graphics #479: Old doc links

jamwaffles, 2020-11-15:
Some old doc links point to `jamwaffles/embedded-graphics/embedded-graphics`. The redirect to `jamwaffles/embedded-graphics/embedded-graphics` works fine, however the `/embedded-graphics` _folder_ no longer exists. 

[This support issue](https://github.community/t/i-want-to-re-name-a-directory-how-do-i-redirect-so-that-my-existing-links-arent-broken/10568) suggests that the redirect/alias should work, but in my testing on `raw.githubusercontent` URLs it doesn't. Maybe it doesn't work on the raw domain, I dunno.

One option to try is a symlink like `ln -s ./embedded-graphics .` which may get old links working again.

Similar to #462.

rfuest, 2022-03-25:
Should we close this? I don't think there is a good way of fixing the broken images in the older versions of the e-g docs and newer versions use inlined images.

jamwaffles, 2022-03-27:
Yeah I think we can close this. As time and versions continue to increase, it will become even less relevant as well.

closed 2022-03-27 by jamwaffles