embedded-graphics/embedded-graphics #480: Improve fonts module documentation

rfuest, 2020-11-15:
From https://github.com/embedded-graphics/embedded-graphics/pull/470#pullrequestreview-530789384:
> One thing that's missing that I think is important is an example of implementing a custom font. I don't know if it's possible to add one that's short enough to show in the docs, but it would really help tie all the traits and structs together.
>
> There's a reference to the text-custom-font simulator example somewhere (I can't find it in the diff now for some reason) which might be worth turning into a link too.

The `text-custom-font` example is barely documented. At least the constants in the `MonoFont` trait should be documented.

Are there any other aspects of the documentation that could be improved?

rfuest, 2021-03-05:
Regarding the `Baseline` enum variants:
> I think these could/should be explained better than just echoing the variant names.
>
> For example I'm not entirely sure what `Alphabetic` refers to, although I have a guess. Also, does Top refer to the topmost pixel of the character, or something else? Perhaps we could adapt the text from the linked canvas docs.

_Originally posted by @bugadani in https://github.com/embedded-graphics/embedded-graphics/pull/561#discussion_r588148338_

bugadani, 2021-03-05:
To clarify, my comment tried to point out an ambiguity with regards to the Top alignment (top of the letter, vs top of the "em box").

cbc02009, 2021-03-07:
This would be very helpful for me. I'm learning embedded rust by trying to program a simple read sensor and print result to display project, and while it works great I'd love to use a font that takes up more of the screen to make it easier to see.

rfuest, 2021-03-07:
The need for a bigger font has come up multiple times in the past. embedded-graphics 0.7 will have support for including BDF fonts, which allows users to include bigger fonts. But perhaps it would be a good idea to include at least one larger font in e-g itself, which would be easier to use.

Which font are you currently using and on what type/resolution of display? And approximately how many times bigger should the font be for your application?

cbc02009, 2021-03-07:
> The need for a bigger font has come up multiple times in the past. embedded-graphics 0.7 will have support for including BDF fonts, which allows users to include bigger fonts. But perhaps it would [...]

I'm using an ssd1306 128x64 display to print three digits of temperature values (e.g. "97.9"). I think something like 40x50 would work well for my use case.  I'm not sure it's a big enough difference from `Font24x32` to add into the eg crate, but documentation would be helpful for figuring it out myself.

rfuest, 2021-03-07:
> I'm not sure it's a big enough difference from `Font24x32` to add into the eg crate,...

The maximum font size was actually reduced in e-g 0.7. `Font24x32` wasn't looking very good, because it was just a scaled up version of `Font6x8`. Our new biggest font is only 10x20 pixels, so that there might be a greater need to add at least one bigger font.

> ..., but documentation would be helpful for figuring it out myself.

I agree that the docs need more work. Did you look at the custom font example for e-g 0.6? https://github.com/embedded-graphics/examples/blob/main/eg-0.6/examples/text-custom-font.rs It might be enough to get you started, but feel free to ask here or in the Matrix channel if you need additional information.