embedded-graphics/embedded-graphics #481: Name for seven segment crate

rfuest, 2020-11-15:
Can anyone think of a good name for the seven segment text renderer crate? The obvious `embedded-seven-segment` is too generic IMO. Maybe we could start a "namespace" for crates related to embedded graphics by using a common prefix, like: `eg-seven-segment` or `e-g-seven-segment`.

In the future I'm also considering to distribute some BDF fonts or font families as crates, which would make them easier to use than having to manually download an include BDF files. IMO it would be nice if they could follow a similar naming scheme, like: `eg-font-comic-sans`.

jamwaffles, 2020-11-15:
`segger` ;) (totally not taken...) `7egger`? `se7en`? Definitely a film you should watch... `eggsegs` (short for e-g segments). Is there a nice German word? Schönessiebensegmentubercraten? I don't have much to add :/ 

> In the future I'm also considering to distribute some BDF fonts or font families as crates

There are two collections [here](https://github.com/spacerace/romfont) and [here](https://github.com/Tecate/bitmap-fonts) which you might also want to distribute. IMO it would be good to keep a collection-oriented approach, as flooding crates.io with a crate for each font (even if namespaced) doesn't feel right to me.

rfuest, 2020-11-15:
> `se7en`? Definitely a film you should watch...

`whats_in_the_box` would be a fitting name for a crate ;).

> Is there a nice German word?

Not really. I don't think that "Siebensegmentanzeige" is a very easy word for non german speaking users.

> IMO it would be good to keep a collection-oriented approach,...

Bundling a huge collection would make it harder to set the correct license for the crate. By bundling font families the corresponding crate could use the same license as the font and the user wouldn't need to check which license applies to which font in the collection.

jamwaffles, 2020-11-16:
As long as it's something about working with the start of a list ;)

> Bundling a huge collection would make it harder to set the correct license for the crate.

Good point, I didn't think of that. Unfortunately the romfonts repo I linked doesn't group well into families, so that would have to be published as like 50 separate crates I suppose. The other repo has 54 families which is also a whole pile of crates.

Generating the crates and releasing them wouldn't be difficult I think - a few bash scripts of `just` commands to loop them all - but I'm still conscious of the noise it would produce, even if "namespaced". That said, I can't come up with a better solution. Perhaps I'm overthinking this...

rfuest, 2020-11-16:
> Good point, I didn't think of that. Unfortunately the romfonts repo I linked doesn't group well into families, so that would have to be published as like 50 separate crates I suppose.

My goal was quality instead of quantity. I would like to provide a few creates with high quality BDF font families to make them easier to use. Other fonts can be included by the user and we can provide some links to sources of additional BDF fonts in the readme.

Most fonts in the `romfonts` repo will be very similar to each other and also similar to the `ibm437` font crate. And the licensing is also unclear.

Should we also update the monospaced fonts that are bundled with e-g? Some of them, like the pixel doubled/quadrupled versions don't look very good IMO. Or should we also publish monospaced fonts as crates?

jamwaffles, 2020-11-16:
I see what you're thinking now, thanks for explaining. A few small quality crates would be great. I also didn't consider that the generic BDF crate can draw any BDF passed to it by the user. 

> Should we also update the monospaced fonts that are bundled with e-g?

I wouldn't be opposed to that. The scaled fonts could at least be scaled using better smoothing algorithms for a start.

rfuest, 2020-11-16:
> I also didn't consider that the generic BDF crate can draw any BDF passed to it by the user.

It includes a proc-macro to include a BDF file, so it's pretty easy to include a custom BDF. But I still think there would be a benefit in having some fonts as crates.

The proc-macro also supports including of subsets, for example:
```rust 
const HELV_B24: BdfFont = include_bdf!(
    "../bdf-parser/tests/u8g2/tools/font/bdf/helvB24.bdf",
    'A'..='Z' | 'a'..='z' | ' '
);
```
I'm not quite sure how this can be replicated if we distribute BDF fonts as crates. But I haven't looked into this yet.

> I wouldn't be opposed to that. The scaled fonts could at least be scaled using better smoothing algorithms for a start.

I hope that we can find a good font, because the result will always be better than an algorithmically scaled font.

jamwaffles, 2020-11-16:
And if we can't, we can always recommend [profont](https://crates.io/crates/profont) for a better solution, but a good set of builtin fonts would be a great feature to have regardless.

jamwaffles, 2020-11-16:
> I'm not quite sure how this can be replicated if we distribute BDF fonts as crates.

Wouldn't the crate just be a thin wrapper around the `include_bdf!` macro?

> I hope that we can find a good font, because the result will always be better than an algorithmically scaled font.

And if we can't, we can always recommend [profont](https://crates.io/crates/profont) for a better solution, but a good set of builtin fonts would be a great feature to have regardless.

rfuest, 2020-11-16:
> Wouldn't the crate just be a thin wrapper around the `include_bdf!` macro?

Yes, ideally the create would only consist of `pub const SOME_FONT_NAME = include_bdf!("some_font.bdf");`. This would make it easy to use the entire font in any project, but you couldn't set the filter anymore. And while const fns can do a lot I don't think it's currently possible to filter an array into another one at compile time.

Some possible solutions would be to add cargo features for common char ranges or a adding a `some_font_name_subset!('a'..='z')` macro to each font crate.

rfuest, 2020-12-07:
I decided to go with `eg-seven-segment` for the seven segment crate. It's not most creative name ever, but at least the purpose of the crate should be clear.

Would you mind if I use the same naming convention for the BDF crate? `eg-bdf` instead of `embedded-bdf`? This would fit in with the `eg-font-...` crates I suggested above.

jamwaffles, 2020-12-07:
Is the BDF crate e-g specific enough to have it in the name? If it has an interface that can be used for parsing without e-g, maybe we should go with `tinybdf` or something without eg in the name?

rfuest, 2020-12-07:
The `eg-bdf` and `eg-bdf-macros` crates are the e-g specific part. The parser you have written doesn't use e-g and can be released as a separate `bdf-parser` crate.

rfuest, 2020-12-07:
We could also split the repo into a parser and e-g specific part, if you prefer that.

jamwaffles, 2020-12-07:
Hmm. Let's do that another time if we need/want to, or someone asks for it. Workspace release management has been a nightmare.

Is the BDF parser available as part of the API, even if e-g is a dependency? That way there's only one crate, but people who just want the BDF parsing bit can still use it.

rfuest, 2020-12-07:
> Is the BDF parser available as part of the API, even if e-g is a dependency?

No, `eg-bdf` works differently than the `tiny...` crates. The BDF file isn't parsed at runtime, which would be inefficient. A proc-macro in `eg-bdf-parser` converts the BDF file into a const array of glyphs. `eg-bdf` doesn't depend on the parser, only `eg-bdf-parser` does.

jamwaffles, 2020-12-07:
Ah ok, I see! `eg-*` sounds good for the BDF naming.

rfuest, 2020-12-07:
Do you think the repo should be renamed to `eg-bdf` as well? IMO that's the main crate in that repo, at least for most users. But there is also `bdf-parser`, which would benefit from a more generic name. (:bike: :house:)

jamwaffles, 2020-12-07:
Up to you. I think I'd leave it as is. I think most people will visit the repos via the readme from links from docs.rs or crates.io, so the readme can explain what the crates are. 

Github's description field could also add a quick explanation for those browsing via the repo list.

rfuest, 2020-12-07:
Thanks for your input! You are right that the repo name doesn't matter for most users, so I'm probably overthinking this. I think I'll keep the name for now, but perhaps I'll rename it to `bdf` without any prefix. That would be most generic and also be fitting for the parser, that isn't an embedded (no_std) project.

I think all other points in this issue have been resolved or added to the tracking issue, so I'll close this now.

closed 2020-12-07 by rfuest

jamwaffles, 2020-12-07:
Glad I could help! Yep, I think we're good for closing this issue now :ok_hand: