embedded-graphics/embedded-graphics #483: Add FrameText

rfuest, 2020-11-16:
Continuation of https://github.com/embedded-graphics/embedded-graphics/issues/469#issuecomment-724960958:
>FrameText would render text inside a box with an optionally colored background. This might sound like it conflicts with what I said in the previous paragraph, but the internal FrameText would be much simpler than what e-t implements and, for example, only support a single line of text. And I think that e-t could benefit from FrameText, because it can reuse the same API to the font renderers.
>
>The digital time display in analog-clock.rs is an example https://github.com/embedded-graphics/simulator/blob/da56ee43c37765c30ac084b1aac91bae782cfb1a/examples/analog-clock.rs#L129 of why this would be useful. Setting the background_color on a Text object doesn't result in a good looking output for most fonts. For this reason the example draws a rectangle (fill_solid) before the text is drawn on top of it (draw_iter). If we would implement a FrameText object the text with background could be drawn using a single fill_contiguous draw call.
>
>I haven't figured out the exact feature set a FrameText implementation would have, but I would like some feedback if this is an idea that is worth pursuing.

jamwaffles, 2020-11-16:
Sounds good to me. Supporting different anchor points would be a good feature to have. It would make the analog clock example work nicely for one example.

Concerning multiline text, I think it would have to support it if the font renderer/e-g internal font did too, otherwise the user would get text overflowing the box in the multiline case.

rfuest, 2020-11-16:
> Supporting different anchor points would be a good feature to have. It would make the analog clock example work nicely for one example.

I agree. We will also need to add padding settings to the frame text styling.

> Concerning multiline text, I think it would have to support it if the font renderer/e-g internal font did too, otherwise the user would get text overflowing the box in the multiline case.

I'm planning to change the font renderer API to render single lines and move the multiline logic to styled `Text`. This way it should be able to restrict `FrameText` to a single line for now. Otherwise we would get pretty close to reimplementing e-t.

Concerting the overflow: Text renderers will be required to clip any text that would be rendered outside the background area. Like `overflow: hidden;`in CSS.

jamwaffles, 2020-11-16:
There might be some overlap with #225 and it's beautiful image I drew so long ago :grin:

rfuest, 2020-11-16:
:rofl:  I just wrote about that issue in #461.

jamwaffles, 2020-11-16:
:cold_sweat: too many issues! aaaaaaa

bugadani, 2020-12-06:
> Otherwise we would get pretty close to reimplementing e-t.

Well, yes, but the current text changes also make e-t hard to update anyway.

jamwaffles, 2020-12-07:
> Well, yes, but the current text changes also make e-t hard to update anyway.

Is there anything missing that you need to update e-t?

> Otherwise we would get pretty close to reimplementing e-t.

Let's not do that :grin: 

> I'm planning to change the font renderer API to render single lines and move the multiline logic to styled Text.

The vision I had for multiline was a very simple "just print some stuff" idea, but maybe it would be best to remove multiline support from e-g and let e-t handle that? As I said in a comment long ago, I can handle positioning a couple of lines of debug text manually if we don't have builtin multiline. Multiline is a bit of a rabbit hole (RTL/LTR, etc) so maybe too complicated for e-g itself?

If we remove it, does `FrameText` serve much purpose?

bugadani, 2020-12-07:
> Is there anything missing that you need to update e-t?

On the technical side, I need to start doing it to see for sure. Moving the text stuff to core would help, but isn't necessary. Font renderers are weird and without pixel iterators e-t needs to be completely redone. I'm fine with that, though, I can see the advantages.

e-t relies heavily on measuring the individual characters and white spaces, I'm not sure how that fits into the current text stuff in e-g.

Apart from technical reasons, I'm missing the time, and I also need to see what you intend to do inside of e-g. FrameText is _the_ reason e-t exists, the fancy styling stuff is icing on that cake. I wonder if it'll be worth maintaining two different text boxes when one is [intended as] a clear superset of the other. I doubt (can't be sure, though) a multiline impl would be much help for e-t as well, since a lot of the fancy stuff probably can't be built on top of that.

> maybe it would be best to remove multiline support from e-g and let e-t handle that

Not including `FrameText` would align with a vision of modularity a bit better, IMO. A text box can be pretty complex, and maybe e-g is big enough without that.
Including `FrameText` would overshadow e-t a lot. I'm not sure if this is a pro or a con, since e-t itself is probably a lot more than what people would need (that can be hidden under feature flags if necessary, though), and it is also not the most beautiful of crates, but that can be remedied.

The decision is ultimately yours. One argument for going with a FrameText is that I don't have a lot of time playing with e-t right now, so I might not be able to pull it together for the 0.7 release. I'm not sure if a synced release is necessary, though.

jamwaffles, 2020-12-14:
With the text alignment stuff in #510, I don't think we should include `FrameText`-like functionality in e-g itself. A `FrameText`-like result can be achieved by drawing a rectangle and setting the text horizontal/vertical alignment without _too_ much boilerplate. 

I think the builtin text capabilities of e-g are about where they should be after #510 (correct me if you have other features planned @rfuest). If a user wants more complex functionality, they should probably reach for e-t at that point.

rfuest, 2020-12-14:
I think I made my reason for adding `FrameText` clear in the initial comment in this issue. But because you both believe that it isn't necessary I'll close this issue now.

closed 2020-12-14 by rfuest