embedded-graphics/embedded-graphics #484: Sector drawing improvements

rfuest, 2020-11-16:
Drawing of sectors has some issues that should be fixed at some point:
- There is no line join in the center:
  [image]
- The lines can overlap the outer edge of the sector, if the stroke width is large relative to the radius of the sector:
  [image]
- 180° degree sectors should produce a straight line, but they sometimes have a 1px offset:
   [image]
- The fill sometimes overlaps the stroke and the background also shows through:
   [image]
  diameter = 50
- Sectors with a sweep angle > 360° should be drawn as a simple circle IMO:
   [image]
- The sector gets flipped around when `angle_start` is exactly 360°:
[image]

jamwaffles, 2020-11-16:
Who'd've thought doing a graphics could be so difficult lol

> There is no line join in the center:

We're lucky I did all the work for thick polylines!

rfuest, 2020-11-16:
> Who'd've thought doing a graphics could be so difficult lol

This was supposed to be a short issue with a single screenshot for the line joins, but somehow that didn't work out as expected ;).

> We're lucky I did all the work for thick polylines!

That should make this much easier. :+1:

jamwaffles, 2020-11-16:
I was thinking of renaming the crate to `embedded-edgecases-and-rabbit-holes-with-maybe-some-pixels` so it's more accurate.

> That should make this much easier.

Some of it, hopefully :)

jamwaffles, 2021-01-20:
I've edited the original post to hide the completed issues. This makes it easier to see what's left at a glance.