embedded-graphics/embedded-graphics #492: Investigate inlining images in docs

jamwaffles, 2020-11-19:
An interesting solution was presented in [this tweet](https://twitter.com/m_ou_se/status/1329192390536597504):

[image]

I wonder if it would be possible to generate, or at least read images out of a folder, and inline them into the docs.

This would help for examples and would fix the stale link issues going forward. It would also be very helpful to describe some of the algorithms in use.

None of this may be feasible, but I thought I'd raise it as an investigation topic.

rfuest, 2020-11-19:
> I wonder if it would be possible to generate, or at least read images out of a folder, and inline them into the docs.

Reading the images out of a folder shouldn't be to hard. We could convert the png files to base64 and use a simple `include_str` in the docs to include them. This wouldn't require any additional build dependencies, if we add the base64 encoded versions to the repo.

Automatically generating images from code in the doc comments would be really nice to have, but I guess it would be a lot harder to implement.

I've found this post which describes a way to use a build script to include images, which might also be worth investigating: https://mrh.io/cargo-build-images-in-rust-docs/

> It would also be very helpful to describe some of the algorithms in use.

Do you mean that in general or by adding images?

jamwaffles, 2020-11-19:
> We could convert the png files to base64 and use a simple include_str in the docs to include them.

That's what I was thinking too.

> I've found this post which describes a way to use a build script to include images, which might also be worth investigating

I wonder if docs.rs supports that though. I guess it should, if the assets are copied to the `target/doc/` folder.

> Do you mean that in general or by adding images?

By adding images :)

rfuest, 2020-11-19:
> I wonder if docs.rs supports that though. I guess it should, if the assets are copied to the `target/doc/` folder.

I did some a quick research before posting the idea and I think it should. The docs.rs provides some information how to test doc builds locally, which would be useful for trying this out.

> By adding images :)

I assume this wouldn't be screenshots in most cases, but something like in https://github.com/embedded-graphics/embedded-graphics/issues/182 or https://github.com/embedded-graphics/embedded-graphics/issues/225. An inline SVG file might also be a possible solution for these kind of graphics.

jamwaffles, 2020-11-19:
> The docs.rs provides some information how to test doc builds locally, which would be useful for trying this out.

It's surprisingly simple, so props to the docs.rs team for making it easy!

> I assume this wouldn't be screenshots in most cases, but something like in #182 or #225. An inline SVG (included with include_str) file might also be a possible solution for these kind of graphics.

Precisely :)

jamwaffles, 2021-06-04:
See also #598