embedded-graphics/embedded-graphics #499: Changing the implementation of styleds

rfuest, 2020-11-22:
To implement a new styled drawable we currently implement, for example, `impl <C: PixelColor> Drawable for Styled<Rectangle, PrimitiveStyle<C>>`. This does only work inside the main e-g crate and makes it impossible to resuse the `Styled` struct in other crates. I think e-t implements its own `Styled`.

I did implement a prototype a while ago, where I added a new `StyledDrawable` trait, which roughly looked like this:
```rust
pub trait StyledDrawable<S> {
    type Color: PixelColor;

    fn draw_styled<D>(&self, target: &mut D, style: &S) -> Result<(), D::Error>
    where
        D: DrawTarget<Color = Self::Color>;
}
```
This makes it possible to implement new styleds in other crates. And I think this change is even non breaking, because `Styled` can be change to use the new `StyledDrawable` impls.

Now to the semi-breaking part of this suggestion: The new `draw_styled` method could also be used directly:
```rust
let style = PrimitiveStyle::with_stroke(BinaryColor::On, 1);

// before:
Line::new(start, end)
    .into_styled(style)
    .draw(target)?;

// after:
Line::new(start, end)
    .draw_styled(target, &style)?;
```
Would we want to advice users to use `draw_styled` instead of the `into_styled` & `draw` combination, if we implement this change? The old code would still work and could be useful to store a styled primitive for a longer time. But for the short lived object, that are often used in e-g code, the new variant might be easier to use. And there might be a slight performance improvement, because the style is passed using a reference and doesn't need to be cloned into a `Styled` object first.

jamwaffles, 2020-11-23:
> I did implement a prototype a while ago, where I added a new StyledDrawable trait, which roughly looked like this:

This would mean there are two different ways to do mostly the same thing, with about the same lines of code. This would be more complex to document and maintain and I'm not sure I see the benefit.

I've typed out multiple half replies to the rest of the post now and nothing has become much clearer, so could use some concrete examples of how this breaking change would fix some problems.

bugadani, 2021-02-19:
> I think e-t implements its own Styled.

Well I missed this, but I don't know how big a deal a simple confirmation is: e-t indeed uses it's own `Styled[TextBox]`. But this didn't really hurt the lib, I think. A TextBox needs it's own style object as well as a custom draw algorithm, so I don't really know what a common Styled trait would add to it. Sure, I also started with "why can't I implement Styled on this thing", but in the end it didn't really matter. But it might just be nonsense from my sleepy brain.

closed 2021-06-05 by rfuest