embedded-graphics/embedded-graphics #501: Migrate tests to new MockDisplay assertion functions

rfuest, 2020-11-23:
All tests, that compare a `MockDisplay` with an expected pattern, should use `MockDisplay::assert_pattern` or `MockDisplay::assert_eq`.

Some suggestion that came up in PR #493 should also be implemented:
* A basic unit test for the output https://github.com/embedded-graphics/embedded-graphics/pull/493#pullrequestreview-535978137
* Macros to allow assertions with messages https://github.com/embedded-graphics/embedded-graphics/pull/493#pullrequestreview-535980029
* A legend for the diff colors in the output https://github.com/embedded-graphics/embedded-graphics/pull/493#discussion_r528191078

rfuest, 2020-11-23:
Do you think it would be useful to remove the `PartialEq` derive from `MockDisplay`? This might help in the migration and we wouldn't accidentally use `assert_eq!`.

jamwaffles, 2020-11-23:
Yes, that would help. We have the `MockDisplay::assert_eq` method which can be used instead so removing the derive sounds fine to me.

PR #508 (2020-11-30): Migrate tests to new MockDisplay assertions
Thank you for helping out with embedded-graphics development! Please:

- [ ] Check that you've added passing tests and documentation
- [ ] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, etc) if your changes affect the **public API**
- [ ] Run `rustfmt` on the project
- [ ] Run `just build` (Linux/macOS only) and make sure it passes. If you use Windows, check that CI passes once you've opened the PR.

## PR description

This PR changes all tests to use the new `MockDisplay` assertions.

Ad [... cut]
closed 2020-11-30 by rfuest