embedded-graphics/embedded-graphics #505: Finish crate split

rfuest, 2020-11-29:
From https://github.com/embedded-graphics/embedded-graphics/pull/498#pullrequestreview-540484691:

- [ ] Move `PointsIter` from into `embedded_graphics_core::iterator`
- [ ] Move `RawDataIter` into `embedded_graphics::iterator`
- [ ] Check which types in the `iterator` module need to be in `core`
- [ ] Try to remove draw target adapters, like `Clipped` or `ColorCoverted`, from `core`

rfuest, 2020-11-29:
How will we document which e-g-core version is compatible with which e-g versions? Should a README file of an external crate mention the e-g-core version it's compatible with instead of the e-g version? The core version number might not be familiar for application developers who only depend on e-g itself.

rfuest, 2020-11-29:
I just noticed that we missed to move the external font renderer support to the core crate.

jamwaffles, 2020-11-29:
Dangit. What needs moving over though? I can see the `Text` struct and `TextStyle`/`TextStylePixels` traits, but I'm not sure what else.

rfuest, 2020-11-30:
`TextStyle` and `TextStylePoints` should be in core and `Text` should stay in e-g. Additional traits might be added in the future to add frame text support.

jamwaffles, 2020-11-30:
Both `TextStyle` and `TextStylePixels` (which is what I assume you mean by `TextStylePoints`) require `Text` though, so we'd have to move `Text` into core as well.

rfuest, 2020-11-30:
Good point. I'm not sure yet if it's better to move `Text` into core or replace `Text` by `&str` and `Point` in `TextStyle` and `TextStylePixels`.

jamwaffles, 2020-11-30:
I'm not sure either. At any rate, moving the text stuff into core won't be a breaking change as far as I can see, so we can do it at our leisure.

Maybe @bugadani can add some comments on what's required from e-g-core in terms of text.

bugadani, 2020-11-30:
I think TextStyle and co. is rather renderer-specific, so I wouldn't necessarily move them. To be fair, I haven't had the time to dive into 0.7 and how it affects e-t. In the current versions, although I use TextStyle and TextStyleBuilder, they aren't strictly necessary. I treat e-t as it's own text renderer, and for that the most important dependency is the Font trait, besides the mechanisms to create my own pixel renderers.

I think I should spend some time with the new code, though, before I say anything legally binding.

rfuest, 2020-11-30:
> I think TextStyle and co. is rather renderer-specific, so I wouldn't necessarily move them.

`TextStyle` was changed to be a trait and isn't renderer specific. The old `TextStyle` is now called `MonoTextStyle`.

bugadani, 2020-11-30:
Interesting. Since it works with `Text` so closely, I don't *think* extracting it would make a difference for e-t, though. Kinda hard to describe a textbox using only a position :)

rfuest, 2020-12-01:
#509 removes the `TextStylePixels` trait and replaces the `Text` arguments in the `TextStyle` methods, so that `Text` can stay in `embedded-graphics`.

bugadani, 2021-01-03:
After porting e-t over to use e-g 0.7, I need to depend on the main crate instead of core for a number of reasons:
 - e-t implements the `Transform` trait for `TextBox` and co., which (the trait) is not present in `core`.
 - Currently e-t supports `MonoFont` by default. I could make this optional, behind a feature flag for example.
 - Current tests and examples are built on top of `MonoFont` support. I'll need to extend this to other font renderers, though.

If e-g core would include the `Transform` trait, e-t could be split into a renderer-agnostic core (some more cores and we'll have a cob of corn in our hands!), as well as a complete package with support for the most widely available font renderers.

rfuest, 2021-01-08:
I would prefer to not add additional traits to the core crate at and have e-t depend on the main e-g crate instead. Because we control e-t it shouldn't be too hard to keep it compatible with the latest e-g version.

The `Transform` trait might change in the future and shouldn't be included in `core` at the moment IMO. Perhaps we could include it in a future version of e-g-core, if we recognize that it's necessary, but we should try to keep the core crate as small as possible.

bugadani, 2021-01-08:
All right, depending on e-g proper works as well.

jamwaffles, 2021-04-19:
Bumping this to see if we can close it or there's any more to do.

rfuest, 2021-04-21:
I'll take another look and close it if I don't find anything more that needs to be done.

jamwaffles, 2021-05-15:
It looks like the two unchecked items in the OP are no longer valid. The discussion now seems to be around text items which could be added to core as their own piece of work in the future, likely as a non-breaking change. With this in mind, I'll close this issue and we can open news ones in the future for other work.

closed 2021-05-15 by jamwaffles