embedded-graphics/embedded-graphics #511: Tracking issue for text rendering changes

rfuest, 2020-12-02:
- [ ] Add `FrameText` #483
- [ ] Imprt BDF code into `embedded-bdf` repo
- [ ] Fix text examples
  The changes in PR #510 will break the examples
- [ ] Add text alignment example
- [ ] Use `FrameText` in analog clock example
- [ ] Update builtin monospaced fonts or add additional fonts as separate repo
- [ ] Try to distribute BDF fonts as crates
- [ ] Add simple external font renderer example to e-g docs

bugadani, 2021-01-04:
I've been thinking about strikethrough and underlined text support in e-g. Because fonts can look whatever, just dashing a 1px line through them is suboptimal, and so this should be supported on a per-renderer basis. This would remove the responsibility out of e-t to try and guess these decorations, which I consider a good thing. This also means I don't have to knock on your doors to publish MonoFont's pixel iterator :) 

For MonoFont, at least for 6x8 it means the following:
 - For underlined text to look good, underlined characters are 1 px taller than otherwise
 - We could add the decoration programatically, as it is currently done in e-t. This works for 6x8 but may require multiple lines for bigger fonts to look good, something I haven't checked yet. This poses the problem of describing the decoration in the font data.
 - We could add 3 (ul, st, ul+st) new glyphs for the fonts and overlay those during rendering. Depending on the font, the underline may require making the glyphs taller, and I'm not sure what would be the nicest solution here - 6x9 glyphs in the 6x8 font sounds fishy, after all.

rfuest, 2021-01-08:
I agree that underlining and strikethrough can't be handled generically for all text renderers and therefore should be handled by the individual renderers.

IMO strikethrough should be implemented differently: Instead of changing some pixels to `text_color` during the glyph rendering I would draw the line after the glyph is rendered. This would cause some overdraw, but has two advantages IMO: Most text doesn't use strikethrough, so we wouldn't need to slow down rendering of normal text by the additional strikethrough logic. And if we add alpha support later it would be possible to draw a semi transparent strikethrough with the text visible behind the stroke.

To make the semi transparent strikethrough possible we would need to add color settings for strikethrough and underline, which would be a nice to have feature anyway IMO. Perhaps as a custom enum: `None`, `TextColor`, `BackgroundColor`, `Custom(C)`.

bugadani, 2021-01-08:
Is transparent strikethrough a common feature we should support? For me it sounds esoteric enough for us to ignore, we are targeting resource-constrained systems, after all. Third party renderers may do as they wish, especially if it's as complicated as drawing a line over the text, so extending MonoFont wouldn't probably be terribly difficult here.

You make a good point with the overhead.

A bit different question that came up:
`e-t` currently renders on a per-character basis. Since font renderers are only required to provide a `render_line(..., &str)` to draw text, I need to rework how e-t draws text.

This is mostly fine, as I can take substrings and render those. However, e-t supports fully justified alignment, which means whitespace width isn't constant, so I can't simply call `render_line(" ")` for those. While I can probably work around this using a clipped draw target, if renderers could provide a way to fill a space with whatever, it would be a big help. If we ever want to do text decoration, `e-t` can't just use a rectangle, unfortunately.

rfuest, 2021-01-08:
> Is transparent strikethrough a common feature we should support?

No, IMO strikethrough itself is already pretty uncommon. But if we draw the line after the glyph this would just work if we add alpha support.

> This is mostly fine, as I can take substrings and render those. However, e-t supports fully justified alignment, which means whitespace width isn't constant, so I can't simply call `render_line(" [...]

Why would you need a clipped draw target in this case? Couldn't you call `render_line` for each word? Perhaps we should rename the method to `render_text` and add another method to get the line height instead of returning it from `render_line`.

bugadani, 2021-01-08:
> Why would you need a clipped draw target in this case?

I need to draw the whitespace if the user sets background color, and strikethrough/underline should also be rendered there. I can do this by repeatedly drawing a single space, but if the justified layout calculates that a space should be `1.25` characters wide, well I need to clip a whole space somehow :)

rfuest, 2021-01-08:
Ah, that makes sense. Would an additional `draw_whitespace(point, width)` method, that draws the background and lines, work for e-t?

bugadani, 2021-01-08:
Yes, I believe it would work.

rfuest, 2021-01-10:
@bugadani Do you think that we could make e-t completely renderer agnostic?

I think it would help to split the `TextBoxStyle` into two styles instead of duplicating all settings in `MonoTextStyle` or similar types from other renderers. This would make the API a bit more complicated to use, but would cleanly split character from paragraph styling, which is common in word processors and DTP applications. To help with this split I would suggest to move the horizontal and vertical alignment settings from `MonoTextStyle` to `Text`, because they aren't used by e-t and don't belong to the character styling.

bugadani, 2021-01-10:
@rfuest certainly. Work has already been started in https://github.com/embedded-graphics/embedded-text/pull/90 - the PR is probably not even close to what it should look like, but it's an attempt at least in this direction.

For `MonoFont` and the likes, the task is actually not that difficult. There are only a few requirements and so far embedded-text did all the other work:
 - Measuring a character
 - Drawing a character
 - Drawing whitespace

The current string-based rendering provided by e-g is actually a lot more flexible, where the font renderer itself handles special characters (like `\r` or a soft hyphen). It could also make support for fancier features, like connecting cursive characters, possible. The core of e-t isn't up to this task, currently, as I need to change the implementation to work in terms of strings instead.

There are a lot of open questions if we are to make e-t into something of a general package. How does escape sequence styling affect text rendering? Take for example a cursive font renderer that connects characters - what if, inside a word, someone changes font color? e-t will need to issue two different draw calls, how will it communicate the connection between characters? I'm quite sure we need a time-generality tradeoff somewhere.

I'm not sure what you mean by "horizontal alignment ... aren't used by e-t". The alignment settings aren't the same, but they are certainly used. At any rate, e-t never relied upon `MonoTextStyle`'s alignment settings to it seems fine to me.

Long story short:
 - I'm working on it
 - I'm trying to make use of whatever e-g provides (assuming e-g will implement `draw_whitespace` :) ).
 - We'll probably need several iterations in the future on how the interface will need to look.

rfuest, 2021-01-10:
I'm currently working on moving the text renderer API to the core crate. This first version should contain everything that is required to render simple fonts. We will probably need to extend or change the API when we want to support advanced features, like ligatures, in the future, but I don't want to over complicate the first version.

> I'm not sure what you mean by "horizontal alignment ... aren't used by e-t".

That e-t didn't use the settings in `MonoTextStyle`. The `MonoTextStyle` vertical alignment is different from the e-t version, because it only sets the position of the first line and not the vertical alignment of the entire paragraph (I'm considering to change the name to `baseline` to match the related CSS property). But IMO these settings shouldn't be present in `MonoTextStyle` if they are ignored by e-t.

> I'm working on it

:+1: 

> I'm trying to make use of whatever e-g provides (assuming e-g will implement `draw_whitespace` :) ).

I'm working on it :wink:

> We'll probably need several iterations in the future on how the interface will need to look as I don't want to be the reason to delay e-g 0.7 a lot longer.

I agree that this will probably need tweaks later, but I think we should get some basic version ready for the 0.7 release, whenever that will be.

bugadani, 2021-01-10:
> But IMO these settings shouldn't be present in MonoTextStyle if they are ignored by e-t.

Agreed.

Currently, e-t wraps the font's style into the textbox's style. Perhaps e-g should do something similar to that with Text. I believe this is exactly what you proposed, so I'm just stating the obvious, am I not? :)

rfuest, 2021-01-10:
> Currently, e-t wraps the font's style into the textbox's style. Perhaps e-g should do something similar to that with Text. I believe this is exactly what you proposed, so I'm just stating the obvious, am I not? :)

I'm not quite sure about the exact implementation yet, so it might not be totally obvious ;). At the moment I'm considering to add the alignment options directly to the `Text` struct instead of wrapping the style inside another style. `Text::new("abc", pos)` would create a text with the default alignement and `Text::with_alignment("abc", pos, h_algin, v_align)` could be used set another alignment.

bugadani, 2021-01-10:
I think I'd prefer a style for that - maybe even if it means one for the font, one for the "target". I would like to see `Text` and `TextBox` be similar in this regard and `TextBox` certainly has more options, which I wouldn't like to (re)do as a huge array of constructors :) Maybe `Text` will have some as well in the future, who knows?

rfuest, 2021-01-10:
OK, that makes sense. I'll try to implement this in my next PR.

rfuest, 2021-01-10:
I've opened an draft PR for the new version of the text renderer API. It should contain most changes required for e-t, but it still requires some polish and needs to be tested with other renderers.

rfuest, 2021-01-11:
The BDF font renderer is now partially ported to the new text renderer API: https://github.com/embedded-graphics/embedded-bdf/pull/5 It still need some work, but the basics work, without having to implement any special drawables, like before.

bugadani, 2021-02-10:
Just wanted to chime in that the recent changes cover everything necessary to bring embedded-text up to date with embedded-graphics. Still quite a lot of work to be done on my end, but at least now the project can compile and passes the tests using latest alphas.

rfuest, 2021-02-10:
That sounds great! I'll try to not break the API too many times until the 0.7 release ;)

ryankurte, 2021-02-16:
not sure this is the right place for it but, related to breaking text rendering API changes and y'all may already have a plan to solve this?

have y'all considered dynamic bounds on the font / styles etc., rather than the type parameters on everything? at the moment if you want to say, render in more than one style, you have to do rebuild the style / everything else because they're technically different _types_, where they could instead take a reference to an impl of the relevant trait (`Font`, for example)?

also in few places, like alignment, these types would appear to be representable as an enum rather than a type/trait to simplify the container further? in some instances it's possible to write container enums to mitigate this, however, where associated constants are used this becomes impossible. The `Font` trait is a particularly tricky candidate as the associated constants would appear to preclude loading of external fonts (as they need a container type with matching constants).

as an example, to abstract over multiple font sizes, `ts`, `tb`, 'a', 'v', and 'h' are distinct types thus cannot be stored in the same place, so one must exponentially expand this for all possible fonts -and- styles (alignments etc.):

```
FontKind::Font8x16 => {
                let ts = TextBoxStyleBuilder::new(Font8x16)
                    .text_color(BinaryColor::On)
                    .height_mode(h)
                    .alignment(a)
                    .vertical_alignment(v)
                    .build();

                let tb = tb.into_styled(ts);

                tb.draw(&mut self.display).unwrap();

                tb.size()
            }
            FontKind::Font12x16 => {
                let ts = TextBoxStyleBuilder::new(Font12x16)
                    .text_color(BinaryColor::On)
                    .height_mode(h)
                    .alignment(a)
                    .vertical_alignment(v)
                    .build();

                let tb = tb.into_styled(ts);

                tb.draw(&mut self.display).unwrap();

                tb.size()
            }
... repeat for every font and style you may wish to render
```

bugadani, 2021-02-17:
Extensibility is one aspect - if embedded-graphics were to provide a `Fonts` enum, I don't see how third parties could define their own fonts. Same is true for alignments, "height modes" (an example is the `Scrolling` alignment that was added to the set, it's possible there are other useful algos that aren't currently implemented in the crate).

A known limiatation in embedded-text is that while in principle it could render text with different fonts at the same time, it's not easy to describe that on the type level.

I suppose it's possible to group `Font`s together into `FontFamily` enums and define a font renderer for those - however, I'm not sure how we should do that currently.
An idea I'm considering would be to implement a FontFamily as an n-tuple of fonts, but I'll need some experimentation to see what's even possible.

This is definitely something I want to do for embedded-text, but requires time to come up with a solution that won't need to rearchitect the world one more time (ideally, not changing embedded-graphics at all).

rfuest, 2021-02-17:
I've been recently working on a draft PR that converts the `MonoFont` trait into a struct. This would make all monospaced fonts have the same type. This should help with some of these problems.

> also in few places, like alignment, these types would appear to be representable as an enum rather than a type/trait to simplify the style container further?

This only applies to embedded-text, the internal `TextStyle` already uses enums for the alignment settings. I would also prefer to use enums in both cases to remove some generics and provide a more consistant API in e-t and e-g.

rfuest, 2021-02-17:
Do you expect that other crates will extend the formatting options in e-t or why do you think that an enum based solution will be less extensible?

bugadani, 2021-02-17:
Once I was convinced the alignment options were done, I got a request for a new one. I know I can't think of everything :) But I actually have an idea how to work around this. Or maybe I could just go doen the route of feature requests. Anyway, consistency would be a plus here.

rfuest, 2021-02-17:
Adding a `non_exhaustive` attribute to the enums might be a good idea. This would make adding new alignment options non breaking changes.

ryankurte, 2021-02-17:
> I've been recently working on a draft PR that converts the MonoFont trait into a struct. This would make all monospaced fonts have the same type. This should help with some of these problems.

A `MonoFont` struct that is consistent for all the monospace fonts would be excellent, perhaps paired with a `Font` trait that this implements for rendering? This would simplify the use of monospace stuff, while ideally allowing others to implement different fonts / renderers.

Rather than an enum for all the fonts, a dynamic trait binding`&dyn Whatever` in place of generics would mitigate this as you could then refer to static `Font` instances (which actually sorta makes sense in this context I think) without changing the object -types-, or wrap a bunch of fonts in an enum and re-implement the trait over it if this was desirable, and you can still use dynamic dispatch without `Box` without it being an issue.

> Once I was convinced the alignment options were done, I got a request for a new one. I know I can't think of everything :) But I actually have an idea how to work around this. Or maybe I could just go doen the route of feature requests. Anyway, consistency would be a plus here.

One option is to maintain the alignment trait, implemented over an enum rather than a set of types in the library, and use that as a default type bound so unless you're using something custom the types are all consistent, but if you want something custom you can drop that in instead?

> Out of curiousity, could you describe your use case, please?

I'm trying to use it in a sorta customisable / composable manner, basically exposing a higher-level API for rendering out a pages / windows / window-stacks, but, so long as the type signatures differ (and the traits are non-genericisable) it's proving to be a struggle. My hope is to replicate some of the higher-level composition that [pebble](https://web.archive.org/web/20161201053328/https://developer.pebble.com/docs/c/User_Interface/Window_Stack/) used to have with WindowStacks (which was the best setup for embedded GUIs I have ever come across) and I long-ago [partially replicated](https://github.com/ryankurte/micro-gui/blob/master/example/source/menu_window.c#L97) in C.

rfuest, 2021-03-04:
Sorry for the late reply.

> ...perhaps paired with a `Font` trait that this implements for rendering?

e-g 0.7 adds a `TextRenderer` trait, which can implemented for text style objects to add support for other text renderers. We have started to implement two external renderers for BDF fonts and a scalable seven segment font: https://github.com/embedded-graphics/bdf and https://github.com/embedded-graphics/eg-seven-segment

> ...but the traits would need to be reworked to be dispatch-able (no associated constants).

Do you have an idea how we could make the `TextRenderer` trait object safe? Here is an abbreviated version of the trait:
```rust
pub trait TextRenderer {
    type Color: PixelColor;

    fn draw_string<D>(
        &self,
        text: &str,
        position: Point,
        baseline: Baseline,
        target: &mut D,
    ) -> Result<Point, D::Error>
    where
        D: DrawTarget<Color = Self::Color>;
}
```
The associated type shouldn't be a problem, because we can `&dyn TextRenderer<Color = C>` where the trait is used. But I don't see a good way to get rid of the generic type `D` on the drawing function. If `DrawTarget` would be object safe we could also use dynamic dispatch for the `target` parameter, but we couldn't use custom error types anymore:
```rust
pub trait TextRenderer {
    type Color: PixelColor;

    fn draw_string(
        &self,
        text: &str,
        position: Point,
        baseline: Baseline,
        target: &mut dyn DrawTarget<Color = Self::Color>,
    ) -> Result<Point, ???>;
}
```

And making `DrawTarget` object safe would mean that iterators would be passed by using dynamic dispatch, which will probably have a higher performance impact (just an untested assumption).

> Also please feel free to tag / message in other places if I can be of any help in designing / reviewing any of these things, I have a _wee bit_ of experience with designing traits / abstractions as a maintainer on embedded-hal, and this is an awesome library / effort!

Thanks for that offer, it's really appreciated.

ryankurte, 2021-03-04:
> Sorry for the late reply.

nw! life is busy / fully expect the same from me eh.

> Do you have an idea how we could make the TextRenderer trait object safe? Here is an abbreviated version of the trait:

how about _inverting_ this perhaps? then it doesn't _need_ to be object safe but also could be. Generally the draw target and colour will be constant for a given system, much more so than the font / text, so associated types for target / colour types, and I think with colour extracted the font may not even need to be dynamic dispatched because the function call will just get monomorphized at the call site.

something like:

```rust
impl <E, D, C> TextRender for D 
where
  D: DrawTarget<Error=E>,
  C: Colour,
  E: Debug,
{
  fn draw_string<F: Font>(
        &mut self,
        text: &str,
        position: Point,
        baseline: Baseline,
        font: &F,
        colour: &C,
    ) -> Result<Point, E>
}
```

The fonts traits should (imo) still be made object-safe, which could be via swapping to methods to grab the associated constants and having the `TextRender` impl do the rendering (including converting from bits to colours so fonts don't need to be generic over colour), and there might be cases in which dynamic would help but, it's hard to know without playing with em.

> And making DrawTarget object safe would mean that iterators would be passed by using dynamic dispatch, which will probably have a higher performance impact (just an untested assumption).

I believe the performance implications of dynamic dispatch are not overly significant, at least with c++ the vtable perfomance was sometimes surprising, but, it seems this may or may not actually be required depending on the font implementation.

ryankurte, 2021-03-04:
> ... so the font would provide an api that lets it set bits for a given character that is then translated via the renderer. that might take some playing with...

oh, this would pretty much just be `Drawable` but simpler / mediated by the Display/TextRender impl rather than taking a `Display` object.

rfuest, 2021-03-05:
The idea of implementing a `TextRender` for `DrawTarget`s is interesting and I'll play with it.

> (including converting from bits to colours so fonts don't need to be generic over colour, which makes sense i think as fonts are not different per-colour?)

I think I left out some important details in my abbreviated version. `TextRenderer` is implemented for text style object. These can contain additional colors, e.g. as a background color or to underline the text.

> i believe the performance implications of dynamic dispatch are not overly significant, at least with c++ the vtable perfomance was sometimes surprising, but, it seems this may or may not actually be required depending on the font implementation.

That might be the case. One of my goals for e-g 0.8 is to do more testing/benchmarking on actual target hardware to get a better idea about the performance of e-g.

jamwaffles, 2021-04-19:
Is there anything left to do on this ticket? I'm a bit tired to read through the comments but it looks like the remaining TODO items can either be moved to new issues as future feature additions (larger font, integer scaling, etc) or marked as complete (improve docs).

bugadani, 2021-04-20:
@rfuest Right now it looks like everything but `TextStyleBuilder` implements `From<Style>`. Was `TextStyle` left out of this intentionally, or is this something that was missed?

rfuest, 2021-04-21:
> Is there anything left to do on this ticket? I'm a bit tired to read through the comments but it looks like the remaining TODO items can either be moved to new issues as future feature additions (larger font, integer scaling, etc) or marked as complete (improve docs).

I think you are right. I'll move the future features to other issues and close this one.

rfuest, 2021-04-21:
> @rfuest Right now it looks like everything but `TextStyleBuilder` implements `From<Style>`. Was `TextStyle` left out of this intentionally, or is this something that was missed?

No, this was unintentionally missed.

jamwaffles, 2021-05-15:
I'm going to close this ticket, as most of it has been implemented/discussed at this point.

I've opened #593 and #594 to track the remaining items.

Thanks for the contributions everyone!

closed 2021-05-15 by jamwaffles