embedded-graphics/embedded-graphics #515: Organization membership visibility

rfuest, 2020-12-07:
I did change the visibility for my e-g organization membership to public a while back. If you aren't a member of the org the page for the e-g org now only shows my avatar:
[image]

This makes me look more important than I am :wink: Should I change my membership visibility back to private or do @jamwaffles and @bugadani also want to set their visibility to public?

jamwaffles, 2020-12-07:
All I do round here is polish @rfuest's socks and make coffee but...

I've changed my visibility to public anyway!

I'd suggest @bugadani do the same :)

bugadani, 2020-12-07:
Thanks for the vote of confidence, now I can wear the membership with pride 😅

jamwaffles, 2020-12-07:
The 3 musketeers are now public! Closing...

closed 2020-12-07 by jamwaffles

rfuest, 2020-12-07:
Thanks for making my avatar feel less lonely on the org page!