embedded-graphics/embedded-graphics #516: Keeping debugging tools

rfuest, 2020-12-07:
During the development of features we often create temporary examples that help us to develop that feature. I did, for example, create one for the recent segment changes that allows you to change the angles and stroke width using the keyboard. These debugging tools aren't good examples of how to use e-g, but IMO it would be useful to start to keep them to make it easier to debug future issues.

Do you think this would be a good idea and if so, where should we put these tools?

jamwaffles, 2020-12-07:
I'd really like this. I've often written pretty complex testing tools for various e-g features which I'd like to keep around for future debugging.

We could make a separate repo, or put them in `examples`, or `build-tools`.

rfuest, 2020-12-07:
> We could make a separate repo,...

That would be possible, if we don't mind to have lots of repos.

> or put them in `examples`,... 

I would prefer not to mix the debug tools and the examples, because they I want to be able to write messy code in the debug tools that no user should take as an example of how to write e-g code. We could hide it in a separate branch in the `examples` repo, similar to `gh-pages`.

> or `build-tools`.

IMO the debug tools don't belong in this repo.

jamwaffles, 2020-12-07:
> That would be possible, if we don't mind to have lots of repos.

I think I'm leaning towards this solution. That would put the total at 10 repositories, which I think is still manageble.

I'd much prefer not to have to maintain two branches in `examples`.

> IMO the debug tools don't belong in this repo.

Agreed, I was testing the waters on that one.

rfuest, 2020-12-07:
> I think I'm leaning towards this solution.

OK, I agree that a separate repo is the cleanest solution. Any suggestions for a name? `debug-tools`?

jamwaffles, 2020-12-07:
Yeah, `debug-tools` is fine. Although, if we want other related stuff in there in the future, maybe `dev-tools` is a broader name. Go with `debug-tools` if you prefer it :)

rfuest, 2020-12-07:
Do you have anything in mind that might be added to `dev-tools` that wouldn't fit into `debug-tools` and also doesn't belong in `build-tools`?

jamwaffles, 2020-12-07:
This is another case of me overgeneralising; I don't :)

rfuest, 2020-12-07:
OK, I'll create a `debug-tools` repo. If there is a compelling reason to change the name we can still rename the repo later, which shouldn't affect many users.

rfuest, 2020-12-07:
The repo is created and the core team should have admin permissions.

rfuest, 2020-12-07:
I've added a basic README so that it should now be possible to create PRs.

closed 2020-12-07 by rfuest