embedded-graphics/embedded-graphics #517: Only Font6x6 is working

rp-, 2020-12-08:
- Version of embedded-graphics in use (if applicable): 0.6.2 and 0.7.0-alpha.2
- Target: ATmega2560 and SSD1306
- If using real hardware: SSD1306

https://github.com/jamwaffles/ssd1306

## Description of the problem/feature request/other
If I use any Font other than `Font6x6` on my specified hardware, my display will not update and it seems also no panic handler is
called (as I tried an serial panic handler). But if I either do not draw text at all, or use the `Font6x6` it will work as expected.

I also ported the SSD1306 driver to the 0.7.0 DrawTarget, but with the same result and additionally there the `Font6x6` is gone.

I'm not sure if it is a bug in the embedded-graphics library or some weird stuff in the SSD1306 driver itself, as I'm pretty new to
embedded programming.

jamwaffles, 2020-12-08:
Do you get a panic if you draw other items to it like primitives or images? It's strange that it's only fonts.

`Font6x6` is removed in 0.7.0-alpha.2, as can be seen in [the changelog](https://github.com/embedded-graphics/embedded-graphics/blob/master/CHANGELOG.md#removed-1).

The embedded Rust ecosystem is very new on AVR. Embedded-graphics itself has only been targetting 32 bit and higher CPUs so far, so I wouldn't be surprised if there were issues with smaller word sizes. That said, I don't think it should cause a panic. 

See if you can get your panic handler printing some useful information - maybe you're running out of memory somewhere. At any rate, it would be useful to know exactly what is causing issues.

rp-, 2020-12-08:
I can draw a `Rectangle` just fine, but didn't try other primitives. SetPixel also works.

Maybe also interesting, as soon as I use `Text::new` I also have to disable `overflow-checks` in the debug build otherwise I get the following linker error:

```
error: linking with `avr-gcc` failed: exit code: 1
  |
  = note: "avr-gcc" "-mmcu=atmega2560" "-Wl,--as-needed" "-L" "/home/rp/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/avr-atmega2560/lib" "/home/rp/work/coding/avr-mega2560-gfx/target/avr-atmega2560/debug/deps/avr_mega2560_gfx-3dcd632a039de8f2.ssd1306-14a3298c9c7d8432.21nkcox9jks071cj.rcgu.o.rcgu.o" "-o" "/home/rp/work/coding/avr-mega2560-gfx/target/avr-atmega2560/debug/deps/avr_mega2560_gfx-3dcd632a039de8f2.elf" "-Wl,--gc-sections" "-no-pie" "-L" "/home/rp/work/coding/avr-mega2560-gfx/target/avr-atmega2560/debug/deps" "-L" "/home/rp/work/coding/avr-mega2560-gfx/target/debug/deps" "-L" "/home/rp/work/coding/avr-mega2560-gfx/target/avr-atmega2560/debug/build/ssd1306-10b11638824ebd46/out" "-L" "/home/rp/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/avr-atmega2560/lib" "-Wl,-Bstatic" "/home/rp/work/coding/avr-mega2560-gfx/target/avr-atmega2560/debug/deps/libcompiler_builtins-ba0b09ebaf0021be.rlib" "-Wl,-Bdynamic" "-lgcc"
  = note: /usr/bin/avr-ld: /home/rp/work/coding/avr-mega2560-gfx/target/avr-atmega2560/debug/deps/libcompiler_builtins-ba0b09ebaf0021be.rlib(compiler_builtins-ba0b09ebaf0021be.compiler_builtins.n6sj7f6l-cgu.7.rcgu.o): in function `compiler_builtins::int::shift::Ashl::ashl':
          /home/rp/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/vendor/compiler_builtins/src/int/shift.rs:(.text._ZN17compiler_builtins3int5shift4Ashl4ashl17hc9f985f3fdaa0d1fE+0x150): undefined reference to `core::panicking::panic'
          /usr/bin/avr-ld: /home/rp/work/coding/avr-mega2560-gfx/target/avr-atmega2560/debug/deps/libcompiler_builtins-ba0b09ebaf0021be.rlib(compiler_builtins-ba0b09ebaf0021be.compiler_builtins.n6sj7f6l-cgu.7.rcgu.o): in function `compiler_builtins::int::shift::Lshr::lshr':
          /home/rp/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/vendor/compiler_builtins/src/int/shift.rs:(.text._ZN17compiler_builtins3int5shift4Lshr4lshr17hbb03edb8ec40a8beE+0x1f8): undefined reference to `core::panicking::panic'
          /usr/bin/avr-ld: /home/rp/work/coding/avr-mega2560-gfx/target/avr-atmega2560/debug/deps/libcompiler_builtins-ba0b09ebaf0021be.rlib(compiler_builtins-ba0b09ebaf0021be.compiler_builtins.n6sj7f6l-cgu.1.rcgu.o): in function `compiler_builtins::int::specialized_div_rem::u32_div_rem':
          /home/rp/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/vendor/compiler_builtins/src/int/specialized_div_rem/norm_shift.rs:(.text._ZN17compiler_builtins3int19specialized_div_rem11u32_div_rem17h36cd71f48d8fe7f1E+0x718): undefined reference to `core::panicking::panic'
          collect2: error: ld returned 1 exit status
```

Yeah and I don't get a panic on the serial, I tried a manual panic and that would show on the serial output, so the handler seems to work.

rp-, 2020-12-08:
I did more testing and I have some success also with the Font6x12 (0.7.0-alpha2), as it is the smallest size of all `.raw` files.
And so think that the ATmega2560 just runs out of memory :/

I don't know if you have some possible memory optimizations there, the plane .ELF file seems to be ok.

```
AVR Memory Usage
----------------
Device: atmega2560

Program:   11266 bytes (4.3% Full)
(.text + .data + .bootloader)

Data:       1943 bytes (23.7% Full)
(.data + .bss + .noinit)
```

But I don't see how much stack/heap will be allocated, while running.

rfuest, 2020-12-08:
AVRs are using a Harvard architecture and cannot directly access the flash memory content in the same way as the RAM. This means that constants will probably all be loaded into RAM at startup and the large font files will take up a huge percentage of the RAM.

I haven't done any AVR development in Rust yet, but there seems to be a crate to use constants in flash: https://github.com/cryptjar/avr-progmem-rs

closed 2021-01-24 by jamwaffles