embedded-graphics/embedded-graphics #526: Voids in polylines

jamwaffles, 2021-01-22:
## Description of the problem/feature request/other

In testing #471, I found another issue with polyline drawing: 

[image]

Some pixels around the inner joint aren't drawn.

@rfuest Should we block 0.7 on this?

## Test cases

### Thin line

[image]

```rust
#[test]
fn polyline_voids_1() {
    let mut display = MockDisplay::new();

    let points = [Point::new(5, 30), Point::new(11, 19), Point::new(16, 10)];

    Polyline::new(&points)
        .into_styled(PrimitiveStyle::with_stroke(BinaryColor::On, 3))
        .draw(&mut display.translated(Point::new(-4, -9)))
        .unwrap();

    display.assert_pattern(&[
        "           # ",
        "          ## ",
        "          ###",
        "         ### ",
        "         ### ",
        "        ###  ",
        "        ###  ",
        "       ###   ",
        "       ###   ",
        "      ###    ",
        "     ####    ",
        "     ###     ",
        "    ###      ",
        "    ###      ",
        "   ###       ",
        "   ###       ",
        "  ###        ",
        "  ###        ",
        " ###         ",
        " ###         ",
        "###          ",
        "  #          ",
    ]);
}
```

### Thick line

[image]

```rust
#[test]
fn polyline_voids_2() {
    let mut display = MockDisplay::new();

    let points = [Point::new(5, 30), Point::new(11, 19), Point::new(16, 10)];

    Polyline::new(&points)
        .into_styled(PrimitiveStyle::with_stroke(BinaryColor::On, 18))
        .draw(&mut display.translated(Point::new(3, -6)))
        .unwrap();

    display.assert_pattern(&[
        "          #                 ",
        "          ###               ",
        "         ######             ",
        "         ########           ",
        "        ###########         ",
        "        #############       ",
        "       ################     ",
        "       ##################   ",
        "      ##################### ",
        "     #######################",
        "     ###################### ",
        "    ####################### ",
        "    ######################  ",
        "   #######################  ",
        "   ######################   ",
        "  #######################   ",
        "  ######################    ",
        " #######################    ",
        " ######################     ",
        "######################      ",
        " #####################      ",
        "   ##################       ",
        "     ################       ",
        "       #############        ",
        "         ###########        ",
        "           ########         ",
        "             ######         ",
        "               ###          ",
        "                 #          ",
    ]);
}
```

rfuest, 2021-01-23:
How common is this problem? Does it only exist for almost parallel lines?

> @rfuest Should we block 0.7 on this?

I don't think so. This problem can also be fixed in a minor release. But I don't think that 0.7 will be released any time soon, so perhaps there will be an opportunity to fix it before the 0.7 release.

jamwaffles, 2021-01-23:
Yes, it mostly happens with nearly-parallel lines. I agree that it's not a huge issue and doesn't need to block 0.7.

> But I don't think that 0.7 will be released any time soon

How so? Development has moved slowly recently for sure, but are there any major pieces of work blocking release (aside from #523 which I'll look at asap - sorry for the delay)?

rfuest, 2021-01-24:
> How so?

Mainly the lack of progress at the moment and a lack of motivation on my side. The text rendering changes are the main blocker and this will require additional PRs after #523 to get the API right. I would also like to release a first version of the BDF renderer with 0.7. And #505 also needs to be finished, so that we hopefully don't have to break the core crate shortly after the 0.7 release.

jamwaffles, 2021-01-24:
> and a lack of motivation on my side.

Likewise :c hopefully that will change soon so we can get 0.7 over the line :)

Aside from the text rendering PRs, are there any other breaking changes you can see that would block 0.7? It might be a good idea to save up further breakages for 0.8 so we don't burn out on 0.7.

rfuest, 2021-01-24:
> Aside from the text rendering PRs, are there any other breaking changes you can see that would block 0.7? It might be a good idea to save up further breakages for 0.8 so we don't burn out on 0.7.

I'll need to take another look at the open issues to answer this. I think there were some minor breaking changes I would like to include in 0.7.

rfuest, 2021-01-24:
Back to the original topic of this issue: Might this gap only happen for `Colinear` joins? This was my impression when I played with the debugging tool, but maybe there are other cases which also show gaps.

jamwaffles, 2021-01-25:
That was also my finding. I guess we'll know better once the cause of the issue is found, but I couldn't produce the glitch in another other than nearly straight lines.