embedded-graphics/embedded-graphics #533: New monospaced fonts

rfuest, 2021-01-27:
Should we replace our current set of `MonoFont`s by the new ones in PR #532 or should we keep both?

The conversion tool currently creates a ASCII subset of each font. This keeps the bitmap data small and should be enough for most english texts. I would suggest to also generate a ISO 8859-1 (aka Latin-1) subset of each font so that more users can use the builtin fonts for their language. We could add additional encodings if there is demand for it in the future or refer users to the BDF renderer, which supports arbitrary subsets of fonts.
How should we organize the fonts if we include multiple versions of each fonts? We could add a suffix/prefix to each font (e.g. `Font6x8Ascii`) or add submodules for each subset to the `mono_font` module (e.g. `mono_font::ascii::Font6x8` and `mono_font::iso8859_1::Font6x8`)

jamwaffles, 2021-01-28:
Replacing them with the new ones would simplify things a lot. I also really like the idea of having a "source of truth" for the fonts that isn't some random bitmap image on the internet.

Each approach has its pros and cons, but I think submodules would work better. It reduces the number of unique exports (although does duplicate them under submodules), and allows applications to more easily migrate between character sets by changing a single import.

rfuest, 2021-01-28:
OK, I'll implement it using the submodules approach.

Do you think it's a problem that the maximum font size of the new fonts would be smaller than the old ones (10x20 vs 24x32)?

jamwaffles, 2021-01-28:
It would be ideal if we could keep the existing sizes and just change the underlying but also. Do you think you could find an OSS 24x32 font somewhere? It would help for migrations and feature set if the font sizes could stay the same. I think I saw a TTF to BDF converter somewhere which might help?

On mobile atm so sorry for lack of research myself.

rfuest, 2021-01-28:
> Do you think you could find an OSS 24x32 font somewhere?

Not sure, but most larger fonts I've seen are proportional and are more suited for the BDF renderer.

But the 24x32 was a scaled up version of the 6x8 font anyway, which didn't look very good and was very inefficient memory wise. We could get the same result by adding a way to scale up fonts when they are drawn. But this is a feature that could be added later, if necessary.

 > I think I saw a TTF to BDF converter somewhere which might help?

I don't believe that there will be more suitable TTF than BDF fonts, but I haven't looked yet. We would also need to consider the font licenses for the bundled fonts.

jamwaffles, 2021-01-29:
> We could get the same result by adding a way to scale up fonts when they are drawn. But this is a feature that could be added later, if necessary.

Sounds good to me. We could use a similar approach to [this](http://www.technoblogy.com/show?3AJ7) to support efficient integer scaling for all fonts.

closed 2021-02-02 by rfuest