embedded-graphics/embedded-graphics #536: FYI: Getting Started with v0.6.2 Simulator Examples

follower, 2021-02-01:
Thanks for your work on this crate!

When trying to get started with trying out the crate today I got a bit caught out with the current repo "renovations" leaving the docs & repos not quite in agreement. :D

So, in case it's helpful to others until the dust settles from a non-alpha v0.7.0 release, here's what I needed to do to get the examples (for current crate release v0.6.2) running in the simulator locally (on Linux/Ubuntu/ElementaryOS):

----

* First, do this:

     ```
     $ pkcon install libsdl2-dev # or `apt install libsdl2-dev` if preferred.

     [...]

     $ git clone https://github.com/embedded-graphics/embedded-graphics.git

     [...]

     $ cd embedded-graphics
     ```

* Next, *don't* do this, because if you follow the `README.md` from [master version](https://github.com/embedded-graphics/embedded-graphics/blame/a441b7148af814433e4bc6a8f0f9dfd1ffaef504/README.md#L107), you'll get:

     ```
     $ cargo run -p embedded-graphics-simulator --example hello
     error: package(s) `embedded-graphics-simulator` not found in workspace `/home/philip/Development/embedded-graphics`
     ```

*  So, *instead* you now need to do ([~via](https://github.com/embedded-graphics/embedded-graphics/blame/embedded-graphics-v0.6.2/embedded-graphics/README.md#L164)):

     ```
     $ git checkout embedded-graphics-v0.6.2

     [...]

     $ cd simulator/

     $ cargo run --example hello-world

     [...]
     ```

*  Or, alternatively, it also seems this variant works from the repo root directory:

     ```
     $ cargo run --package embedded-graphics-simulator --example hello-world
     ```

*   Both of which should then eventually give you a graphical hello world.

----
----

- Version of embedded-graphics in use (if applicable): [v0.6.2](https://crates.io/crates/embedded-graphics/0.6.2)
- Target: [simulator] / ~~[real hardware]~~
- ~~If using real hardware: [display driver used]~~

## Description of the problem

The steps to get started with the examples/simulator are not clear due to issues with documentation & in-progress repository split/reorganization.

It seems the Cargo "workspace" & `embedded-graphics-simulator`/`simulator` name patch was removed here: https://github.com/embedded-graphics/embedded-graphics/pull/459/files#diff-2e9d962a08321605940b5a657135052fbcef87b5e360662bb527c96d9a615542L1

In addition to the package `embedded-graphics-simulator` seemingly no longer being in the workspace; the example is also named `hello-world` not `hello`.

(Also, the examples [have now been moved](https://github.com/embedded-graphics/examples/issues/1) to <https://github.com/embedded-graphics/examples>.)

----

Hopefully someone else can save a bit of time with this list of steps. :)

Thanks again & good luck for the ongoing v0.7.0 development!

rfuest, 2021-02-03:
Thanks for this writeup!

@jamwaffles Should we pin this issue until 0.7 is released?

jamwaffles, 2021-02-03:
Good idea! I've pinned it with the existing title, but it should probably be changed to make it clearer that it's for 0.6.x whilst 0.7 is in the works.

jamwaffles, 2021-06-16:
I've unpinned this issue now that 0.7.0 is out and I think we're good to close it too.

closed 2021-06-16 by jamwaffles