embedded-graphics/embedded-graphics #539: Update README for 0.7 release

rfuest, 2021-02-03:
The README needs some updates for the 0.7 release:
- Add links to new font renderer crates (if they are finished for 0.7)
- Update instructions how to run simulator examples (#536)
- Change font attribution
- Link to new migration guide

Please add any additional items that I forgot.

jamwaffles, 2021-02-05:
The only thing I can see right now:

* Migration guide for 0.6 -> 0.7

rfuest, 2021-04-02:
The "Shapes and text" example should be updated:
- The border around the display can be drawn by using `display.bounding_box()` and inside stroke alignment.
- The centered text can use `TextAlignment::Center` and e.g. `display.bounding_box().center() + Point::new(0, dy)`
- In the README version of the example it contains a line that starts with `#`

rfuest, 2021-04-06:
How should we handle the remaining images in the README/doc root? I don't think that we can inline them, because GH strips data URLs from Markdown documents. But the images aren't essential to understand the docs, so that IMO it's not a big problem if the aren't visible if someone views the docs without being connected to the internet.

But perhaps we should change all links to GH to include a specific revision of the repo, so that they remain valid even if we change the repo layout or remove an image from the docs in the future.

jamwaffles, 2021-04-06:
> that IMO it's not a big problem if the aren't visible if someone views the docs without being connected to the internet.

Agreed. I think we should check the alt text of all the images and make sure they're accurate for cases like that - as a reader it's useful to know what I'm not missing if the image didn't load.

> But perhaps we should change all links to GH to include a specific revision of the repo, so that they remain valid even if we change the repo layout or remove an image from the docs in the future.

Excellent idea

rfuest, 2021-04-06:
Do you have an idea how we could make the link to the drawing examples stand out more?

The mysterious YT algorithm had suggested me this video some time ago: https://youtu.be/PfchkNNmonU?t=4095 I did watch some excerpts where they are looking at the e-g docs, which was interesting to get an outside view. At the linked time they don't seem to not notice that the drawing examples still exist in the 0.7 docs.

jamwaffles, 2021-04-06:
We could reshuffle the `lib.rs` docs and move the `Examples` section up the page, but I like the ordering of the rest of the content as is.

Perhaps we could add a shorter `Examples` section just below the list of features at the top which links further down the page to the actual `Examples` section. It could also be used as a place to link to both the `examples` mod in the docs and <https://github.com/embedded-graphics/examples> without using up too much space.

PR #583 (2021-04-19): Bump readme for 0.7
Thank you for helping out with embedded-graphics development! Please:

- [x] Check that you've added passing tests and documentation
- [x] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, etc) if your changes affect the **public API**
- [x] Run `rustfmt` on the project
- [x] Run `just build` (Linux/macOS only) and make sure it passes. If you use Windows, check that CI passes once you've opened the PR.

## PR description

Update the readme as per #539. I'm not going to shuffle the sections around as the last commen [... cut]
closed 2021-04-19 by jamwaffles