embedded-graphics/embedded-graphics #541: New core and e-g alpha release

rfuest, 2021-02-03:
IMO the core crate should be near the final state for 0.7. @jamwaffles do you think it's time for a new core version release and a new e-g alpha? We might need to release another core version before 0.7, but I would like to get rid of all the git dependencies.

jamwaffles, 2021-02-03:
Sounds good to me! As I see it, the new releases would be:

* Core 0.2.0 (because breaking changes)
* e-g 0.7.0-alpha.3

However:

1. We could release core 0.1.2 as I don't think anyone is using it yet
2. I assume we'd want to update e-g to point to core 0.2.0 before releasing it's new alpha, right?

rfuest, 2021-02-03:
> We could release core 0.1.2 as I don't think anyone is using it yet

I think we should increase the version number according to the semver standard (to 0.2.0) regardless if it's used or not.

> I assume we'd want to update e-g to point to core 0.2.0 before releasing it's new alpha, right?

:+1:

jamwaffles, 2021-02-03:
> I think we should increase the version number according to the semver standard (to 0.2.0) regardless if it's used or not.

I agree, I just wanted to see what your thoughts were.

I'll go and do a new 0.2.0 core release now.

jamwaffles, 2021-02-03:
0.2.0 and 0.7.0-alpha.3 are released

closed 2021-02-03 by jamwaffles