embedded-graphics/embedded-graphics #548: Add return type to Drawable::draw

rfuest, 2021-02-12:
`Drawable::draw` currently returns `Result<(), D::Error>` and doesn't allow to return any value. This is OK for most draw calls, but a return value would have an advantage for some.

Here is an example how this feature could be used to chain text drawing with different styles:
```rust
let next_point = Text::new("Label ", Point::new(10, 20)).into_styled(label_style).draw(&mut display)?;
Text::new("Value", next_point).into_styled(value_style).draw(&mut display)?;
```
By using the `next_point` value the size of the first text doesn't need to be calculated twice. This will only work for left aligned text, but for everything else we have e-t.

jamwaffles, 2021-02-15:
That could be useful indeed. Would the impl look something like this:

```rust
pub trait Drawable {
    /// The pixel color type.
    type Color: PixelColor;

    type Output;

    /// Draw the graphics object using the supplied DrawTarget.
    fn draw<D>(&self, display: &mut D) -> Result<Self::Output, D::Error>
    where
        D: DrawTarget<Color = Self::Color>;
}
```

(and once [associated type defaults](https://doc.rust-lang.org/unstable-book/language-features/associated-type-defaults.html) are stable would make upgrading a lot less noisy for most users)

Aside: presumably this will happen after 0.7?

rfuest, 2021-02-15:
Yes, it would look like this.

> Aside: presumably this will happen after 0.7?

`Drawable` is in the core crate and I would prefer to add it before the 0.7 release to make it less likely that we need to break the core crate for 0.8.

PR #552 (2021-02-16): Add output associated type to Drawable
Thank you for helping out with embedded-graphics development! Please:

- [x] Check that you've added passing tests and documentation
- [x] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, etc) if your changes affect the **public API**
- [x] Run `rustfmt` on the project
- [x] Run `just build` (Linux/macOS only) and make sure it passes. If you use Windows, check that CI passes once you've opened the PR.

## PR description

Perhaps a very little interpretation of #548, but here's my go at a return  [... cut]
closed 2021-02-16 by jamwaffles