embedded-graphics/embedded-graphics #553: Support for draw Japanese and Chinese `Text`

SKTT1Ryze, 2021-02-19:
I try to draw some Japanese or Chinese words on my waveshare e-paper, writing codes below:  
```Rust
use embedded_graphics::{
    fonts::{Font12x16, Text},
    prelude::*,
    text_style,
};
use epd_waveshare::{
    color::*,
    epd2in13_v2::{Display2in13, EPD2in13},
    graphics::{Display, DisplayRotation},
    prelude::*,
};
let mut epd = EPD2in13::new(&mut spi, cs, busy, dc, rst, &mut delay).unwrap();
// Setup the graphics
let mut display = Display2in13::default();
    
display.set_rotation(DisplayRotation::Rotate90);

// Draw some text
Text::new("中文!", Point::new(1, 1))
    .into_styled(text_style!(
        font = Font12x16,
        text_color = Black,
        background_color = White
    ))
    .draw(&mut display).unwrap();
        
// Transfer the frame data to the epd and display it
epd.update_frame(&mut spi, &display.buffer()).unwrap();
epd.display_frame(&mut spi).unwrap();
epd.sleep(&mut spi).unwrap();
```
But I get some `?` on the e-paper.  
Does this crate support drawing Japanese or Chinese `Text`?  
If not, how can I add this support?  
Thanks.

SKTT1Ryze, 2021-02-19:
I have found answer in the source code, thanks.

closed 2021-02-19 by SKTT1Ryze

ohmyarch, 2021-04-22:
@SKTT1Ryze So is this problem solved?

SKTT1Ryze, 2021-04-22:
Not yet. It's much more difficult than I thought, bucause we need chinese character library, witch much lager than English one.  
Embedded systems have not so much memory space.

rfuest, 2021-04-22:
The upcoming support for BDF fonts could be a solution for this problem. It allows the user to import a subset of glyphs in a font, so that you don't need to store all Chinese characters in flash, if the used characters are known at compile time.

Any input on what features are missing to support additional languages/writing systems is always appreciated, because we only have a limited knowledge of the requirements for other writing systems.