embedded-graphics/embedded-graphics #554: Add methods to reset settings in builders

rfuest, 2021-02-24:
`MonoTextStyleBuilder` doesn't allow to reset `text_color`, `background_color`, `underline_color` or `strikethrough_color` to a transparent color. This isn't a problem when the builder is used to create a new style, but doesn't allow the builder to be used to modify an existing style: `MonoTextStyleBuilder::from(&old_style).text_color(...)`.

I didn't have time to check the other builders, but there will probably be other instances which are also affected by a similar problem. We should make sure that the all builders provide methods to generate any object regardless of the initial values. And the API should be similar for all builders.

jamwaffles, 2021-02-27:
I agree there should be some symmetry between all the builders. Do you have an idea in mind of what the API should look like? I was thinking something like `.default_<thing>()` or `.reset_<thing>()`? I'm also wondering if there's any utility in adding a `.reset()` method which would reset the builder to its default state, but I guess that would be handled by `ThingBuilder::new()` anyway.

rfuest, 2021-02-27:
> I agree there should be some symmetry between all the builders. Do you have an idea in mind of what the API should look like? I was thinking something like `.default_<thing>()` or `.reset_<thing>()`?

No idea, I haven't looked into this. I think we need to make a list of `<thing>`s that require a reset method. For the colors remove_` prefix would also work, but the name isn't as universal as `default_` or `reset_`.

> I'm also wondering if there's any utility in adding a `.reset()` method which would reset the builder to its default state, but I guess that would be handled by `ThingBuilder::new()` anyway.

IMO `new` and a `Default` impl should be enough.

PR #576 (2021-04-06): Add reset methods to builders
This PR adds the reset methods to builders, that were discussed in #554.

Closes #554
closed 2021-04-06 by rfuest