embedded-graphics/embedded-graphics #562: Changing the behavior of Dimensions for transparent drawables

rfuest, 2021-03-04:
Implementations of the `Dimensions` trait for drawables are currently expected to return the smallest bounding box that contains all drawn pixels. This means that transparent drawables are expected to return a zero sized rectangle.

But this might not be the desirable behavior if you want to layout drawables by using the result of `Dimensions::bounding_box`. An application could toggle the visibility of an element setting the colors to transparent without affecting the layout, if we always return the same bounding box.

The changed `Dimensions` trait would also be easier to implement for `Text` drawables, which have never returned the correct bounding box. `Text::new("    ", ...)...` would need to return an empty bounding box (if the background is transparent), which would be hard to add to the text renderer API.

And I think this would make `Dimensions` trait more consistent in general, because it's also used for `DrawTarget`s, where it always returns the same bounding box, regardless of the display content.

rfuest, 2021-03-04:
cc @bugadani This would probably affect e-t

jamwaffles, 2021-03-04:
Hm that does make sense. If we make this change, do we need to add some functionality to test whether an item is completely transparent? We have some stuff internally which might be enough and external code can always check the style properties but that seems error prone.

bugadani, 2021-03-04:
I'm practically useless today, will read/respond tomorrow.

rfuest, 2021-03-04:
> If we make this change, do we need to add some functionality to test whether an item is completely transparent?

I think that's a good idea. As a first step we could make `PrimitiveStyle::is_transparent` public and add the same method to the other styles, like `MonoTextStyle`.

We could also add an additional method that replicates the current behavior for cases where this is easy to implement. But I'm not sure this is required, because checking `style.is_transparent()` should be easy enough in most cases.

rfuest, 2021-03-04:
> I'm practically useless today, will read/respond tomorrow.

OK, no hurry. I'll also leave PR #561 open until you have reviewed it, because it will break e-t.

jamwaffles, 2021-03-04:
> As a first step we could make PrimitiveStyle::is_transparent public and add the same method to the other styles, like MonoTextStyle.

Sounds good to me. As always, better to be conservative and add stuff later when people ask for it/need it.

> We could also add an additional method that replicates the current behavior for cases where this is easy to implement. But I'm not sure this is required

I don't think so either, and if it turns out it is later, it's easier to add things than take them away.

So to summarise:

* The only breaking change will be to styled bounding box calculations: they will return the same size whether the style is fully transparent or not.
* We'll make `is_transparent` public and add the same method to `MonoTextStyle`. I don't believe there are any other places this needs to be added?

bugadani, 2021-03-05:
e-t shoulnd't be affected that much here. The dimensions of the textbox depend on some style settings, but basically depend on what the user specifies, or how tall the text is, ignoring any pixel colors. Or if I'm wrong, it's how I intended and anything else is a bug.

For some of the text-based height adjustments this was inconsistent with e-g and it doesn't stand on firm foundations, but I just went with this options.

> An application could toggle the visibility of an element setting the colors to transparent without affecting the layout

An application can always choose not to call `.draw()`. Changing the style is a bit iffy because then care must be taken to restore the original settings, and also because the different style objects we currently have (i.e. no standardized `Paint` object that we use for literally everything - **should we have one?**) makes implementation a bit harder.

Although in #439 we went with the old approach, I always found it somewhat weird for the size of an object to change depending of the colors of the paint it's drawn with. "Affecting the layout" is something of a weird point IMO: are we mixing responsibilities here? e-g isn't a UI library, it shouldn't be concerned with layout that much. For a UI library built on top of e-g, I think both approaches are equally valid, though I wouldn't expect a lot of views being rendered with invisible content where the user expects nothing :)

This proposed change would allow layout engines to be more flexible. The thing that comes to mind is Android's `View.VISIBLE`, `View.HIDDEN`, `View.GONE` trio where the three options affect how a particular view is measured, displayed and IIRC even affects interactivity. Now, since we don't (yet) have a layout engine like that, or at least we don't know about them, the proposed change would allow a similar flexibility. I'm all for that. Maybe it's time to dust off embedded-layout and turn it into a proper measure/layout/draw cycle layout engine.

So to summarize:

 - I like this proposal - color shouldn't affect an object's size
 - e-t shouldn't be affected by that much, and if it is, it's probably because it's brokens somewhere currently
 - We should make a UI library and a layout engine

rfuest, 2021-03-05:
> So to summarise:
>
> * The only breaking change will be to styled bounding box calculations: they will return the same size whether the style is fully transparent or not.
> * We'll make `is_transparent` [...]

This would be the plan. `MonoTextStyle` would be the only other place in e-g to need a `is_transparent` method, but similar methods should be added to the other text renderers and e-t.

rfuest, 2021-03-05:
> "Affecting the layout" is something of a weird point IMO: are we mixing responsibilities here? e-g isn't a UI library, it shouldn't be concerned with layout that much.

e-g isn't a layout library, but IMO the `Dimensions` trait should provide useful information to implement layout externally. And I think that the new definition results in more predictable results for the user and is also easier to implement for drawables.

bugadani, 2021-03-05:
> e-g isn't a layout library, but IMO the Dimensions trait should provide useful information to implement layout externally. And I think that the new definition results in more predictable results for the user and is also easier to implement for drawables.

We definitely agree on this point. What I wanted to say was that we shouldn't decide how to hide something, i.e. the transparent = Size::zero() choice probably didn't belong in e-g anyway. I hope I make sense, but in the end I'm just trying to say why I think this proposal is a good one.

rfuest, 2021-03-05:
I just tried to implement this and encountered a problem:
How should we handle transparent strokes? Currently setting the `stroke_width` to `0` or the `stroke_color` to `None` has the same effect. Should `stroke_width != 0 && stroke.color.is_none()` affect the bounding box?

There is also a difference in how transparent colors are handled in `PrimitiveStyle` vs `MonoTextStyle`. Setting the `stroke_color` to `None` simply removes the stroke entirely, but setting `text_color` to `None` renders the text as a `background_color` rectangle with text shaped transparent holes. Should we implement the same for `PrimitiveStyle`?  This would mean that the fill would be reduced in size for `stroke_width != 0 && stroke_color.is_none() && stroke_alignment != Outside `.

bugadani, 2021-03-05:
> Should stroke_width != 0 && stroke.color.is_none() affect the bounding box?

That would be somewhat consistent with my line of thought, so I'd say yes. Whatever option affects size when drawn should also apply when transparent. In essence, I'd treat "e-g skipping transparent pixels" as something of an optimization, but we could just as well send those to the display driver so that it can discard them. I think a pixel is a pixel regardless of the color.

bugadani, 2021-03-05:
> Should we implement the same for PrimitiveStyle?

After arguing with myself for a good 15 minutes, I think yes, simply because the result would be more flexible. Just like in the case of transparent text on colored background (where not drawing the characters is a no-op, thus probably not really useful), stroke should "punch through" the fill when transparent (since not doing that would be a similar no-op). I think doing it this way would be more flexible, at least in terms of what is possible to draw.

jamwaffles, 2021-03-05:
I've written and deleted a pile of answers to this thread :grin:. So far, nothing useful. Something to consider though:

When we add alpha support, how will the behaviour we choose now change, if at all? Shouldn't we be thinking about this feature with alpha support in consideration? 

Conceptually, do we want to treat `None` as the same as "alpha of zero"? If so, then yes a non-zero stroke width should change the bounding box size, otherwise probably not.

bugadani, 2021-03-05:
> When we add alpha support, how will the behaviour we choose now change, if at all? Shouldn't we be thinking about this feature with alpha support in consideration?

I think it depends on whether we treat these things (stroke, text, etc.) as being different layers. Transparent text kinda sets a precedent saying "no, it's not a different layer".

> I've written and deleted a pile of answers to this thread

So have I. But doing that helped me resolve some internal conflicts in my thoughts :) Whether my reasoning makes sense outwards is a different question 🙃

jamwaffles, 2021-03-05:
The layer concept is an interesting idea, but I've always seen e.g. "circle with transparent stroke and a fill" as one layer with no overdraw.

> Whether my reasoning makes sense outwards is a different question

At this point I think I should probably chuck my keyboard in the bin as a benefit to humanity...

bugadani, 2021-03-05:
> The layer concept is an interesting idea, but I've always seen e.g. "circle with transparent stroke and a fill" as one layer with no overdraw.

That's what I'm thinking as well. In which case alpha blending doesn't need to complicate things, I think.

> At this point I think I should probably chuck my keyboard in the bin as a benefit to humanity...

No.

rfuest, 2021-03-05:
> ... stroke should "punch through" the fill when transparent

I think this makes sense and I've implemented this behavior in #563.

> But the whole objects would still have its original size, right?

Yes.

rfuest, 2021-03-05:
I agree that we should treat stroke and fill to be on the same layer and not overlap. For the, IMO less common case, that overlapping is desired, the user could still use two separate draw calls. The other way around would be harder to implement for the user.

PR #563 (2021-03-05): Always take stroke_width into account for bounding boxes
This  PR changes the way `stroke_width` is handled if `stroke_color` is `None`. The bounding boxes that are returned by the `Dimensions` trait will ignore `stroke_color`. The `fill_area` will also be reduced regardless of `stoke_color`. See #562 for more details.

Closes #562
closed 2021-03-05 by rfuest