embedded-graphics/embedded-graphics #570: Using #[must_use]

bugadani, 2021-03-11:
I've ran into a footgun with `Transform::translate[_mut]` where I managed to call `translate` instead of `translate_mut`. `translate` has no side effects and returns a new object, so it's a good candidate for `#[must_use]`. Should we sprinkle the API with `#[must_use]`?

rfuest, 2021-03-11:
I think this is a useful addition. We should check for other instances where this could be helpful for the user. `Rectangle::intersection` would be one example outside the transform API.

bugadani, 2021-03-11:
I think it can go on constructors and field accessors, conversion functions (like those of `Angle`) in general - if the result isn't used, what's the point?

On the other hand, I'm not sure if we want to mark a million functions' results as must_use. Helpful, but probably overkill?

rfuest, 2021-03-11:
> ... ,conversion functions (like those of `Angle`) in general - if the result isn't used, what's the point?

Do you mean the `Angle::to_degrees` and `Angle::to_radians` methods? There is no point, but I'm also not sure if the attribute is required in this case. Methods with a `to_` and `into_` are expected to return a value according to the API guidelines. The std library doesn't add a `must_use` attribute to functions like `u16::to_le()`.

> On the other hand, I'm not sure if we want to mark a million functions' results as must_use. Helpful, but probably overkill?

I agree that we need to find a balance with this.

> We could draw a line and only mark the costly functions or whose names are similar enough (`translate` vs `translate_mut`) which could be thought of as somewhat "risky".

I'm not sure that the cost of function calls needs to be considered in this case. If an expensive object is constructed and never use the compiler will probably optimize it out anyway.

I would add a `must_use` attribute to `translate` even if `translate_mut` did not exist. The std library does the same for methods like `u16::wrapping_sub`. The same would apply to the `abs` and `normalize` methods in the `Angle` trait, which should also get a `must_use` tag IMO.