embedded-graphics/embedded-graphics #593: Integer mono font scaling

jamwaffles, 2021-05-15:
As mentioned in https://github.com/embedded-graphics/embedded-graphics/issues/533#issuecomment-769887633, there's an interesting solution to smoothly scaling fonts [here](http://www.technoblogy.com/show?3AJ7).

It would be great to support integer (1x, 2x, 3x, 4x, etc) scaling of the builtin fonts provided by e-g.

This could potentially be expanded to all fonts in the future as some sort of "filter" framework, but for now let's keep it simpler and scope it to any font provided by e-g.

jamwaffles, 2021-05-15:
#511 mentions adding a larger builtin font, however I'd prefer to implement that with this scaling feature rather than adding one in and potentially removing it later once scaling is supported.

rfuest, 2021-05-15:
> #511 mentions adding a larger builtin font, however I'd prefer to implement that with this scaling feature rather than adding one in and potentially removing it later once scaling is supported.

These two points are unrelated. The larger builtin font was never supposed to be a simple scaled up version of an existing font.

jamwaffles, 2024-12-21:
There's an interesting approach here for scaling by 2x: http://www.technoblogy.com/show?3AJ7