embedded-graphics/embedded-graphics #594: Verify underline and strikethrough for bundled fonts

jamwaffles, 2021-05-15:
This is an unchecked item from #511. I'm not entirely sure what it entails. It may already have been verified, in which case this issue can be closed without further work.

jamwaffles, 2021-05-24:
I wrote a simple debug tool in embedded-graphics/debug-tools#11 which outputs this:

[image]

They all look good to me, so we could close this issue unless there's something I should be looking for and aren't.

rfuest, 2021-05-25:
Thanks for taking a look at this and writing a debug tool.  I think it looks good.

One thing that might be worth trying would be to increase the height of the decorations for bold fonts (and maybe also large fonts) to 2px.

jamwaffles, 2021-05-25:
No problem! Should we repurpose this ticket to make/experiment with those changes? If so, feel free to edit the title to match.

closed 2021-06-21 by jamwaffles