embedded-graphics/embedded-graphics #601: Implementing Display for Point and Size

rfuest, 2021-06-04:
Should we implement the `core::fmt::Display` trait for `Point` and `Size`?

The use case I have in mind for this are error messages for assertions. Using the `Debug` output can get noisy and you need to manually print the components if you want a cleaner output:
```rust
// without `Display` impl
assert_eq!(
    point, other_point,
    "point ({}, {}) must be equal to other_point ({}, {})",
    point.x, point.y,
    other_point.x, other_point.y
);

// with `Display` impl
assert_eq!(point, other_point, "point ({}) must be equal to other_point ({})", point, other_point);
```

If we decide to implement this, how should the output look of the `Display` impls look?
`Point::new(1, 2)` -> `1, 2`
`Size::new(1, 2)` -> `1x2` or `1 x 2` or also `1, 2`?

rfuest, 2021-06-04:
Maybe the example wasn't the best, because the assertion already prints the `Debug` representation. But I think that this could still be useful, also in other use cases.

jamwaffles, 2021-06-04:
This could be useful indeed. I don't think you pay any binary bloat unless the item is displayed, right?

It would be useful to have "normal" formatting alongside `defmt` support in #592, although we should be careful about the output format used by each. I think I'd be good with `(1, 2)` for both point and size but I can see the desire to differentiate the two.

rfuest, 2021-06-04:
> This could be useful indeed. I don't think you pay any binary bloat unless the item is displayed, right?

I think so.

> I think I'd be good with `(1, 2)` for both point and size but I can see the desire to differentiate the two.

Should we include the parenthesis in the output? It's easy to add them in the format string, but hard to remove. Or we could use the `alternate` feature: `format!("{}", point)` -> `(1, 2)` and `format!("{:#}", point)` -> `1, 2`

Especially for assertions I prefer the `x` notation for `Size`s:
"size was 4x2 (expected 1x2)" is easier to read than "size was (4, 2) (expected (1, 2))" IMO
But there might be other cases were the `,` notation is better. I don't know which one we should pick.

jamwaffles, 2021-06-04:
> I don't know which one we should pick.

I don't really know either. I think it's somewhat of an arbitrary choice at any rate so perhaps it's best to go with the additive approach and start small with `Point` using `1, 2` and `Size` using `1x2`. We may want to mention in the docs that these formats are subject to change and should not be relied on, but I think that's an unsaid assumption for many crates already.

rfuest, 2021-06-04:
> I don't really know either. I think it's somewhat of an arbitrary choice at any rate so perhaps it's best to go with the additive approach and start small with `Point` using `1, 2` and `Size` using `1x2`.

SGTM.

> We may want to mention in the docs that these formats are subject to change and should not be relied on, but I think that's an unsaid assumption for many crates already.

I'm not sure about this. I always assumed that you should be able to rely on `Display` impls, but not `Debug` impls. This would make a `Display` change a breaking semver change. But I don't expect that we need to change to format soon and don't think this is a big issue.

jamwaffles, 2021-06-04:
All SGTM too :)

PR #602 (2021-06-05): Implement Display for Point and Size
This PR implements the `Display` trait for `Point` and `Size`, as discussed in #601.

I've slightly changed the output for `Size`s: `123 x 456` instead of `123x456`, because it makes the output more readable IMO.

Closes #601
closed 2021-06-05 by rfuest