embedded-graphics/embedded-graphics #608: In-memory display driver?

kpreid, 2021-06-16:
Is there a `DrawTarget` implementation (display driver) which supports writing to an in-memory (or memory-mapped) framebuffer (given `&[u8]`, a bounding rectangle, and perhaps a pixel format)? There is [`MockDisplay`](https://docs.rs/embedded-graphics/0.7.0/embedded_graphics/mock_display/struct.MockDisplay.html), of course, but that's an implementation designed for testing rather than performance. I took a look at the dependent crates for drivers but they all seemed to be hardware-specific.

If not, would such an implementation be a reasonable addition to the `embedded-graphics` crate?

Besides being reusable code (index computation, bounds checking, and implementations of the optional acceleration methods to avoid redundant index computation) for drawing to displays which want this format (if there are any such), it might be useful for applications which would like to draw into images (and then perhaps draw those images to other `DrawTarget`s).

jamwaffles, 2021-06-16:
Hey, thanks for opening the discussion on this. I think embedded-graphics should remain target agnostic and including a framebuffer with e-g itself would go against that. It could certainly be a separate crate to cater for more generic target requirements, however some or all of what you're looking for might already be covered by the [embedded-graphics-simulator](https://crates.io/crates/embedded-graphics-simulator) crate. 

The simulator can be used to generate images if that's what you're looking to do, but it would be great to know what concrete requirements you have that e-g isn't currently fulfilling.

kpreid, 2021-06-16:
> I think embedded-graphics should remain target agnostic and including a framebuffer with e-g itself would go against that.

I was imagining an implementation that is both a small amount of code and sufficiently generic that, in addition to being usable for images or buffered drawing, would be useful to multiple concrete targets that use addressable framebuffers, either because the final destination is in-memory or because they use a local buffer and a flush operation. For examples in existing e-g drivers, [epd_waveshare](https://docs.rs/epd-waveshare/0.4.0/src/epd_waveshare/graphics.rs.html), [st7920](https://github.com/wjakobczyk/st7920/blob/master/src/lib.rs), and [ls010b7dh01](https://github.com/byronwasti/ls010b7dh01/blob/master/src/lib.rs) all draw pixels to in-memory arrays, which could be supported by a common implementation which knows how to do the range checks and index arithmetic.

(Many of these displays have known sizes and thus may benefit from index arithmetic with constant image width; whether the abstract implementation could be compiled to equally efficient code is a question I don't have an answer for handy, but I hope it can given appropriate inlining hints.)

I was thinking that such code could go into e-g on the same basis that it has not-strictly-necessary draw target _adapters_ such as `ColorConverted`, and text drawing conveniences like baseline and alignment; the benefit is not in doing something hard, but in saving people from rewriting the same old basic arithmetic code.

> some or all of what you're looking for might already be covered by the embedded-graphics-simulator crate.

At a glance, the simulator's `DrawTarget` implementation does not implement any of the acceleration functions. That could be fixed, but I'm also concerned that it would generally evolve in an unsuitable direction simply because its purpose is different; it already unconditionally depends on the `image` crate.

> it would be great to know what concrete requirements you have that e-g isn't currently fulfilling.

My own actual application is sufficiently weird that you definitely should not use it as a use case to steer e-g development, which is why my previous comment was arguing in the abstract. It is a game that runs on GPU-having computers (as opposed to resource-constrained `no_std` environments), where I wanted to be able to draw pixelated text and graphics efficiently without implementing my own font renderer and line-drawing routines. I was already using e-g for a complex case where I needed to implement my own DrawTarget; then I encountered a situation where I just wanted to write text into a texture — just a slice of image data — and was hoping to not have to reimplement the very basics of 2D bounds checking and indexing into an array. (I've [done that](https://github.com/kpreid/all-is-cubes/blob/master/all-is-cubes/src/lum/frame_texture.rs#L188) now, without acceleration, since it's not much code; this issue is not at all blocking.)

rfuest, 2021-06-27:
I had previously also considered to add a generic framebuffer implementation to e-g. I agree with @kpreid that a common framebuffer implementation would be useful. By sharing more code between drivers common bugs, like no or incomplete bounds checking, could be prevented.

IMO this feature would be easier to maintain in the future if we include it directly in e-g and not implement it as a separate crate. But a separate crate would allow us to use a newer MSRV and perhaps benefit from const generics.

jamwaffles, 2022-02-22:
@bernii has released a separate crate for framebuffer usage called [`embedded-graphics-framebuf`](https://crates.io/crates/embedded-graphics-framebuf) fwiw.

rfuest, 2022-03-22:
Am I correct that the SH1106 does map bytes into 8 pixel high and 1 pixel wide columns? This would be more specialized than I intended, but it would certainly also help to have such a generic implementation.

I would like to use the same memory layout for the generic framebuffer as `ImageRaw` does. This way it would be very easy to use a smaller framebuffer as an `Image` to use it as a cache for complex drawing operations, to decode compressed images ahead of time, or to apply some image filter.

jamwaffles, 2022-03-22:
> Am I correct that the SH1106 does map bytes into 8 pixel high and 1 pixel wide columns? This would be more specialized than I intended, but it would certainly also help to have such a generic implementation.

That's correct, yeah. SSD1306 (and other SSD130x) uses this same layout too so while it's certainly specialised it's not too uncommon. 

> I would like to use the same memory layout for the generic framebuffer as ImageRaw does. 

Ooh that's smart, nice idea.

---

Thinking out loud a little, I'd also like to see an experiment with a "contiguous pixels" iterator, and perhaps a "contiguous pixel _colour_s" iterator to aid with e.g. batching DMA writes when possible. A bit OT, but would it make sense to have a `embedded-graphics-extras` crate for the various items discussed in this thread? I can't decide if e-g itself is a good place to put them.

rfuest, 2022-03-25:
> That's correct, yeah. SSD1306 (and other SSD130x) uses this same layout too so while it's certainly specialised it's not too uncommon.

Yes, it is common for 1 BPP displays, I've also used LCDs that use the same memory layout. I only meant that the default layout to be compatible with `ImageRaw`.

> Thinking out loud a little, I'd also like to see an experiment with a "contiguous pixels" iterator, and perhaps a "contiguous pixel _colour_s" iterator to aid with e.g. batching DMA writes when possible.

I'm not sure what you mean by "contiguous pixels". But adding support for DMA and also hardware acceleration like DMA2D in STM32s should be considered in the next version of the `DrawTarget` interface. 

> A bit OT, but would it make sense to have a `embedded-graphics-extras` crate for the various items discussed in this thread? I can't decide if e-g itself is a good place to put them.

I would suggest to include the `ImageRaw` compatible framebuffer in e-g and any different memory layout in a separate crate. Are there any other items that would go into `embedded-graphics-extras`?

jamwaffles, 2022-03-25:
> I'm not sure what you mean by "contiguous pixels".

There's a decent explanation of mostly what I mean [here](https://lupyuen.github.io/articles/optimising-pinetimes-display-driver-with-rust-and-mynewt#8d93). I suppose you could describe it as a kind of "inverse" RLE. Like, if you're using `fill_contiguous` and the iterator happens to produce 18 pixels of the same colour that are next to each other, this iterator can batch those up into a single `fill_solid` call. If the display driver has an optimised impl for that, driver performance might improve a bit.

> I would suggest to include the ImageRaw compatible framebuffer in e-g and any different memory layout in a separate crate.

Makes sense :)

> Are there any other items that would go into embedded-graphics-extras?

Probably not enough at the moment. Perhaps a review of the dependent crates on crates.io might find some things that could be brought into the "official" e-g fold.

rfuest, 2022-03-25:
> Probably not enough at the moment. Perhaps a review of the dependent crates on crates.io might find some things that could be brought into the "official" e-g fold.

IMO `display-interface` would be a good candidate to integrate into the `embedded-graphics` organization. It has been adopted by many display drivers and therefore has become an integral part of many e-g based projects. Perhaps it could be merged with some of the ideas in #647 into an `eg-display-hal` crate. Which could also include the specialized framebuffer for 1 BPP displays like the SSD1306.

elpiel, 2022-04-23:
`embedded-graphics-framebuf` seems like a very useful crate.
What I've been looking for is a crate for `canvas` - you can draw things in it and later on draw it on a target at a specific location.

I have something made based upon the MockDisplay, however, I do face a few issues:

- the struct should implement both `DrawTarget` & `Drawable`
- and also have a way to filter out pixes that haven't been drawn onto - this removes the issue of overriding any previously drawn to the display elements that overlap with the canvas.

I would be more than happy to share the code for this "canvas", as I think it can be quite useful for more complex and overlapping drawing.

rfuest, 2022-04-23:
I hadn't considered adding transparency support to framebuffer implementation that will be included in e-g, but that is indeed a useful feature. At some point e-g will gain real transparency support and include the appropriate new color types with an alpha channel. But even if we add transparency at some point there are still use cases which would benefit from an implementation with a simple 1 bit alpha channel which could be used with every color type.

Instead of using the `MockDisplay` approach of using an array of `Option<C>` to store the pixels it would be more memory efficient to use to separate arrays, one for colors and one for the 1 bit per pixel alpha channel. An array of `Option<C>` can use twice as much memory as and array of `C`.

One remaining issue is how the canvas could be cleared. The default `DrawTarget` implementation would only use `C` and calling e.g. `canvas.clear(Rgb888::BLACK)?` would set all pixels to opaque black pixels. I would suggest to add two additional `DrawTarget` implementations, that could be accessed using `alpha()` and `with_alpha()` getters:
```rust
let mut canvas = Canvas::<Rgb565, 128, 128>::new();

// Fill the entire canvas with opaque black pixels.
canvas.clear(Rgb565::BLACK)?;

// Makes all pixels transparent by clearing the alpha channel.
canvas.alpha().clear(BinaryColor::Off)?
// or
canvas.with_alpha().clear(None)?

// By making the alpha channel accessible it would also be
// possible to erase parts of the image.
// This draws a red square with a transparent circle in the middle.
canvas.clear(Rgb565::RED)?;
Circle::with_center(canvas.bounding_box().center(), 32)
    .into_styled(PrimitiveStyle::with_fill(BinaryColor::Off)))
    .draw(canvas.alpha())?;
```

elpiel, 2022-04-24:
I really like the idea of adding alpha channel support in e-g!
I do indeed use an Option, as for how can this support be added... I think the canvas itself can have a special method e.g. `clear_all` that can clear all the pixels. In my case set them to `None`.

Another feature I forgot to mention is adding support for `crop(area: Rectangle)` the idea being that you can draw stuff on the canvas before drawing it on the actual target (or e.g. another canvas) and crop only the part of the canvas you need.
This feature could be very useful if you only want to show, e.g. a bounding box (with border) with text above in the middle and below and you want to show only part of the text above and below, essentially cropping them.

rfuest, 2022-04-24:
> I do indeed use an Option, as for how can this support be added... I think the canvas itself can have a special method e.g. `clear_all` that can clear all the pixels. In my case set them to `None`.

Sure, that would also work. IMO the exact method to clear the canvas including the alpha channel isn't as important as documenting it, because the behavior is different from other `DrawTarget`s.

> Another feature I forgot to mention is adding support for `crop(area: Rectangle)` the idea being that you can draw stuff on the canvas before drawing it on the actual target (or e.g. another canvas) [...]

I think this feature already exist. I am planning to implement `ImageDrawable` and not `Drawable` for the framebuffer in e-g. This has the advantage that you can use `Image` to set the position where the framebuffer is drawn to the `DrawTarget`. And it has the additional benefit that `SubImage`s can be used to draw parts of the framebuffer to a draw target: https://docs.rs/embedded-graphics/latest/embedded_graphics/image/index.html#sub-images.

Using `clipped` is another way to clip drawing operations: https://docs.rs/embedded-graphics/latest/embedded_graphics/draw_target/trait.DrawTargetExt.html#tymethod.clipped

elpiel, 2022-04-24:
RE: Crop/Clip - what i meant was that the Canvas can be cropped and potentially clipped, but the result will be another Canvas.
Since the canvas is also Drawable, you can draw it onto the provided DrawTarget.
Cropped and Clipped don't seem to have this ability.

I'm only mentioning it because, again, I think it might be useful to be able to draw them onto the target.

> This has the advantage that you can use Image to set the position where the framebuffer is drawn to the DrawTarget.

I haven't thought about that but I'll take a look as it sounds reasonable.
Not sure if something like the Canvas should impl ImageDrawable but I will think about it.

rfuest, 2022-04-24:
> RE: Crop/Clip - what I meant was that the Canvas can be cropped and potentially clipped, but the result will be another Canvas.
> Since the canvas is also Drawable, you can draw it onto the provided DrawTarget.
> Cropped and Clipped don't seem to have this ability.

OK, I see what you mean. If you want the clipped canvas to remain also be a `Drawable` the canvas would need to implement another method of clipping. With the existing e-g functions it would only be possible to clip the base canvas for either drawing onto or for drawing it to some other target.

Do you have a use case in mind where it would be necessary to use the clipped canvas both as a `DrawTarget` and a `Drawable`? Or do you just prefer the consistency this provides?

rfuest, 2022-04-24:
I've just opened #673 which contains a first draft of the framebuffer implementation that will be added to e-g.

elpiel, 2022-04-26:
> Do you have a use case in mind where it would be necessary to use the clipped canvas both as a DrawTarget and a Drawable? Or do you just prefer the consistency this provides?

My use case is only for cropped and not clipped.

elpiel, 2022-05-13:
I want to give a little update:
I've managed to clean up most of the API & code but there is still some work to be done. I'll make the repo public soon and share it here.
I want to make a more complete example, however, drawing on a canvas, cropping and later setting its location on the display is done.

I still can't get rid of `alloc`, as I use `vec![None; pixel_count].into_boxed_slice()` with `Box<[Option<C>]>` for the pixels initially (and for `with_default_color` method).
I'd love some feedback on how to remove the Box & slice. I tried a few times using const generics but it quickly falls apart on the `crop`, where I use a `Rectangle` to crop a specific part of the `Canvas`.

[image]

rfuest, 2022-05-14:
Great to see progress on this. I'm looking forward to try it when it's released.

> I still can't get rid of `alloc`, as I use `vec![None; pixel_count].into_boxed_slice()` with `Box<[Option<C>]>` for the pixels initially (and for `with_default_color` method).
> I'd love some feedback [...]

It's hard to give feedback on this without taking a look at the code. What does your `crop` function do exactly? Does it return a new canvas wich is a copy of a rectangular region or does it return a canvas that uses a reference to the original canvas data? It would help if you could share the `crop` function code or at least its signature.

elpiel, 2022-05-14:
@rfuest sure thing, I've already made the repo public. You can find it here:
https://github.com/LechevSpace/embedded-canvas/
Maybe I can make another set of struct using const generics that will remove the need for allocator.
Main drawback is that it might require another Size & Rectangle structs that will use const generics for the cropping.
I'd love some feedback too!

rfuest, 2022-05-15:
OK, now I get it. You'll need to refactor the code quite a bit to get it working with const generics.

The size of the canvas, that is returned by the `crop`, will always need to be `const` and known at compile time. That would be a serious limitation. One solution would be to return a cropped view into the original canvas instead of allocating a new buffer:
```rust
struct CroppedCanvas<'a, C> {
    parent_buffer: &'a mut [Option<C>],
    parent_size: Size,
    area: Rectangle,
}
```
All drawing operations on the `CroppedCanvas` would draw into the pixel buffer of the parent canvas. If a cropped canvas, that doesn't modify the parent canvas, is required the parent canvas can be cloned before it is cropped. But this has the disadvantage that the cropped canvas would use the same buffer size as the parent canvas, which is larger than necessary.

rfuest, 2022-10-14:
`Framebuffer` was added in PR #673.

closed 2022-10-14 by rfuest