embedded-graphics/embedded-graphics #609: Iterating over pixels in `RawImage` and `Image`

jamwaffles, 2021-06-21:
Context: starting [from this message](https://app.element.io/#/room/#rust-embedded-graphics:matrix.org/$6nOEaC4HWJjrwKGiwDDNHhPgHDRgBCBJmlx5f3dbMdQ) by @jacobrosenthal:

> We have all the sexy new from impls for pixels colors, i was hoping to be able to say greyscale a rgb rawimage or image, either wholesale `rawimage.into()` or via a pixel iterator `image.into_iter().map(|p|Gray8::from(p))).collect()` but I cant seem to figure a way, am I missing anything?

Did we miss out adding methods to iterate over the pixels in both a `RawImage` an `Image`? I think these could be added pretty easily, and would be useful methods to have.

rfuest, 2021-06-27:
The need for providing a pixel iterator was deliberately removed from `ImageDrawable`, because it isn't possible to efficiently implement a pixel iterator for all image drawables. `ImageDrawable` could, for example, also be implemented for a vector graphics file.

We had discussed this earlier and decided to add additional traits if these features are required later. One trait could be used to provide an iterator over all pixels and another trait to access individual pixels: `fn pixel(Point) -> C`.

jacobrosenthal, 2021-06-27:
For context I'm basically trying to use eg as the image-rs library. Run
convolution over images etc.

rfuest, 2021-06-28:
I've created PR #612 to add a `pixel` getter trait and implement it for `ImageRaw`s.

> For context I'm basically trying to use(abuse) eg as the image-rs library. Run convolution over images etc.

Interesting. Do you plan to output the result to a `DrawTarget` or generate another image?

jacobrosenthal, 2021-06-28:
I might need an intermediate image to do more work, but that wouldnt be the
goal. The end result would be the DrawTarget

LLFourn, 2024-09-23:
I was looking for something like `Framebuffer::iter_pixels`. I was able to produce it with `RawDataSlice` manually but it was tricky to figure out. It seems like `ImageRaw` and `FrameBuffer` would benefit from helper methods.

rfuest, 2024-09-23:
`GetPixel` trait implementations for `Framebuffer` and `ImageRaw` have been added since this PR was created. It is now possible to iterate over the pixels in an image by using code like this:
```rust
let fb = Framebuffer::<...>::new();
let pixels_iter = fb.bounding_box().points().map(|p| (p, fb.pixel(p));
```
This method of accessing the pixel data is more flexible because you can also mutate pixel colors or access neighbouring pixels to apply effects.

LLFourn, 2024-09-24:
@rfuest This is a more flexible approach. My concern was that it's not quite zero-cost compared to the `RawDataSice` approach. I didn't actually scientifically measure the difference in performance so it might not exist!

rfuest, 2024-09-24:
Yes, there will be some overhead with the solution I suggested. I'm currently working on replacing the code that accesses the raw pixel data in `ImageRaw` and `Framebuffer` and I'm planning to do some analysis of the performance when this work is completed.

`RawDataSlice` will be replaced in the next e-g release. Can you tell me more about your use case to make sure it will still be supported?

LLFourn, 2024-10-01:
@rfuest I'm basically pre-rendering some stuff in a framebuffer and then scrolling it based on user touch/drag gestures. So I need to iter the pixels inside some subsection of the buffer. Now that I look at it maybe I can use the `sub_image` feature to do this more idiomatically. If that works I wouldn't even need to do what I'm doing!