embedded-graphics/embedded-graphics #611: Draw target test suite

rfuest, 2021-06-27:
While I was recently looking at some driver code I noticed some common issues that could easily missed by driver authors. 

One example is an incomplete bounds check by only testing that `x + y * width` is inside the framebuffer and not explicitly checking the x coordinate. This was drawing operations with `x >= with` wrap around and incorrectly modify the following rows.

Automatically testing driver implementations is hard and might not even be possible in cases where the framebuffer cannot be read back. Instead I would suggest that we implement a simple test suite that can be used by driver authors to validate their implementation.

It would run tests like:
* Draw some basic shapes/text.
  Just as an initial test and to make sure the orientation of the display is correct.
* Draw markers to show the bounding box
 To make sure that `{Origin}Dimensions`  is correctly implemented.
* Draw offscreen
  Should be a noop.
* Draw partially offscreen
  This is especially important for `fill_contiguous`, where the overlapping values from the colors iterator must be discarded. 
* Use some very large and very small integer coordinates
  Some drawing code might e.g. use `p.x as u16` which would cause the coordinates to wrap every 65536 pixels.

These are just some examples and the set of tests could be expanded in the future. The documentation for each test would explain the reason the test exists and what the expected output is.

The API could look something like this:
```rust
let tester = DrawTargetTester::new(display);

// Run an individual test:
tester.offscreen()?;

// Or run all tests:
// The function could take a closure to delay the execution between different tests.
// This way the user can adapt this to their target hardware and decide how they want to switch between tests.
tester.test(|| delay.delay_ms(500))?;
// or
tester.test(|| while button.is_high() {})?;
```

Do you think this would be useful?

rfuest, 2021-12-21:
I didn't expect this would be such a bad idea that it would be closed without a comment, but I guess my contributions are no longer appreciated.

jamwaffles, 2021-12-22:
Apologies, was running through things in a couple of spare minutes I had and didn't take the time to reason about the closures.

For the avoidance of doubt, your contributions are very much appreciated Ralf. e-g truly wouldn't be as high quality as it is without your considerable time and effort dedicated to the project.