embedded-graphics/embedded-graphics #613: Binary size increase due to monomorphization

rfuest, 2021-07-07:
While testing some e-g code on real hardware I noticed that some e-g features can cause significant increases in the binary size.

The example application used both e-g `Text` and e-t `TextBox` to display some text. The `mono_font::draw_string` method is one of the largest methods in the project with ~3.5 KB. This method gets duplicated, when `Text` and `TextBox` are used, even if the use the same character style. The reason for this is that `TextBox` wraps the draw target in a `Clipped` adapter, while `Text` uses it directly.

My target only contains 64 KB flash and each unnecessary `draw_string` method takes a large amount of the available flash. To reduce the overhead we will need to use dynamic dispatch for `DrawTarget`s, but that would require us to change `DrawTarget` to be object safe.

bugadani, 2021-08-13:
> The reason for this is that TextBox wraps the draw target in a Clipped adapter

Oh, that's bad.

jamwaffles, 2021-09-30:
Another issue also [came up in chat](https://matrix.to/#/!SfJCDXZbMHXkPovtKL:matrix.org/$SbTzSwKL7v2vBXgeJvm5I6SPOMsilRXY9y3bmyoVKfI?via=matrix.org&via=tchncs.de&via=fik1.net), quoting here for posterity:

> Hi! I've implemented Drawable for a bunch of structs (each in their own crate). I'd like to import them and put the (heterogenous) struct instances in a Vec, then loop over them and call .draw(&mut display) on each with a delay (and other setup) in between.
>
> This doesn't work because Drawable cannot be made into an object ("the trait cannot be made into an object because method draw has generic type parameters").
>
> Is there a way to circumvent the object safety rules, maybe by wrapping things in a struct or some container? Is there some fundamental reason why this can't work? I realize this is probably more of a Rust question than an embedded-graphics one, but would appreciate any help or direction, thanks!

Different end issue, but same root cause I think.

Any thoughts on how to achieve object safety here?

almindor, 2021-09-30:
> Another issue also [came up in chat](https://matrix.to/#/!SfJCDXZbMHXkPovtKL:matrix.org/$SbTzSwKL7v2vBXgeJvm5I6SPOMsilRXY9y3bmyoVKfI?via=matrix.org&via=tchncs.de&via=fik1.net), quoting here for [...]

Something like [this] works but I had to skip the errors because I don't have a way to know what that would be without generics. Not sure if there's a way to get the error "delegated" here.

jamwaffles, 2021-09-30:
You missed out the URL on your `[this]` :)

almindor, 2021-09-30:
> You missed out the URL on your `[this]` :)

fixed

jamwaffles, 2021-11-25:
Note to self: experiment with/benchmark the compile size of inner functions, e.g. https://internals.rust-lang.org/t/inner-function-sugar/15628