embedded-graphics/embedded-graphics #616: Crate no longer compiles on x86_64-unknown-uefi

DianaNites, 2021-07-24:
- Version of embedded-graphics in use (if applicable): 0.7.1
- Target: x86_64-unknown-uefi
- If using real hardware: N/A, Compile issue.

## Description of the problem/feature request/other

I was upgrading a project of mine from embedded-graphics 0.6.2 to 0.7.1, only to discover that it no longer compiles on my target, failing with the following linker error. This project compiled and worked just fine on the previous version.

```
rust-lld: error: undefined symbol: fmodf
>>> referenced by /home/diana/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-graphics-0.7.1/src/geometry/real.rs:204
>>>               libembedded_graphics-c1700c7c1949986e.rlib(embedded_graphics-c1700c7c1949986e.embedded_graphics.86eb5855-cgu.3.rcgu.o):(embedded_graphics::geometry::real::_$LT$impl$u20$embedded_graphics..geometry..real..real_impl..Real$GT$::rem_euclid::h41cb661794b1b4a3)
>>> referenced by /home/diana/.cargo/registry/src/github.com-1ecc6299db9ec823/micromath-1.1.1/src/f32ext/rem_euclid.rs:5
>>>               libmicromath-1c7ed2f7a5002027.rlib(micromath-1c7ed2f7a5002027.micromath.0691bf78-cgu.6.rcgu.o):(micromath::f32ext::rem_euclid::rem_euclid::h64f3ac500cd833a9)
>>> referenced by /home/diana/.cargo/registry/src/github.com-1ecc6299db9ec823/micromath-1.1.1/src/f32ext/div_euclid.rs:6
>>>               libmicromath-1c7ed2f7a5002027.rlib(micromath-1c7ed2f7a5002027.micromath.0691bf78-cgu.12.rcgu.o):(micromath::f32ext::div_euclid::div_euclid::h7c56aa78846885e6)
```

## Test case

This test case should reproduce the failure.

Compile with `cargo build --target x86_64-unknown-uefi -Zbuild-std=core,compiler_builtins -Zbuild-std-features=compiler-builtins-mem`

```rust
#![no_std]
#![no_main]
use core::panic::PanicInfo;
use embedded_graphics::{
    draw_target::DrawTarget,
    mock_display::MockDisplay,
    pixelcolor::{Rgb888, RgbColor},
};

#[no_mangle]
fn efi_main() {
    let mut display = MockDisplay::new();
    display.clear(Rgb888::BLACK).unwrap();
}

#[panic_handler]
fn panic(_: &PanicInfo) -> ! {
    loop {}
}
```

rfuest, 2021-07-24:
This is probably an issue with the toolchain and not e-g itself. Floating point on `no_std` targets can be problematic.

Does the code compile if you enable e-g's `fixed_point` feature?

DianaNites, 2021-07-24:
It worked before, so something between the versions changed that, and as the target is defined in Rust I don't control it. Maybe the bug is in Rust or `compiler_builtins` and it should provide `fmodf`?

Enabling `fixed_point` fixes the compile failure for now, thanks!

I also discovered that compiling with `--release` also worked for some reason, though that appears to only be because of this minimized example and did not apply to my real project.

rfuest, 2021-07-24:
> It worked before, so something between the versions changed that, and as the target is defined in Rust I don't control it. Maybe the bug is in Rust or `compiler_builtins` and it should provide `fmodf`?

`Arc` and `Sector` were added in e-g 0.7, which use floating point for the angles. It is probably related to this: https://github.com/rust-lang/rfcs/issues/2505

If this turns out to be a common problem we could add a `libm` feature to e-g, which would use the floating point methods from the `libm` crate.

> I also discovered that compiling with `--release` also worked for some reason, though that appears to only be because of this minimized example and did not apply to my real project.

Are you using `Arc` or `Sector` in your real project? Otherwise the functions in `real.rs` should be optimized out by the compiler and not cause any problems during linking. Or do you get another (similar) linking error?

DianaNites, 2021-07-24:
> Are you using Arc or Sector in your real project?

Not directly, but something else is probably pulling them in.

> If this turns out to be a common problem we could add a libm feature to e-g

For now `fixed_point` solves the issue. Are there any significant downsides to it that a `libm` feature wouldn't have?

----

Seems like the larger issue is a Rust one, that should ideally be resolved upstream.

Before opening this issue, I also tried bringing in `libm` manually as a dependency to see if it'd resolve the linker issue, and that didn't work.

rfuest, 2021-07-24:
> For now `fixed_point` solves the issue. Are there any significant downsides to it that a `libm` feature wouldn't have?

No. There might be slight differences in the drawn arcs and sectors between the floating point and fixed point implementations of `Real`. But everything else should be the same.

> Before opening this issue, I also tried bringing in `libm` manually as a dependency to see if it'd resolve the linker issue, and that didn't work.

The `libm` crate does only provide a Rust API and not a `extern "C"` style API. I guess you could add something like this to your project to get it to link:
```rust
#[no_mangle]
pub extern "C" fn fmodf(x: f32, y: f32) -> f32 {
    libm::fmodf(x, y)
}
```

DianaNites, 2021-07-24:
I tried something like that, but it revealed many poor linker errors/undefined symbols, so I don't think its something I can easily bring in that way.

jamwaffles, 2021-09-30:
Closing as fixed with the `fixed_point` feature. Please reopen if this problem persists.

closed 2021-09-30 by jamwaffles