embedded-graphics/embedded-graphics #617: Add Line::with_delta constructor

rfuest, 2021-08-06:
An additional `with_delta` constructor would make it easier to create multiple lines with different starting points but the same orientation and length.

```rust
impl Line {
    pub fn with_delta(start: Point, delta: Point) -> Self {
        Self::new(start, start + delta)
    }
}
```

jamwaffles, 2021-08-06:
`Line::polar` (or `Line::new_polar`) might be useful too

bugadani, 2021-08-13:
I find the `delta: Point` a bit weird. I understand e-g lacks a distinct `Vec2d` type, but here it would make the signature cleaner, I think.

jamwaffles, 2021-08-14:
It would, but the behaviour and structure is exactly the same, so it would only really be a type alias. `Point` makes more sense in most other areas of e-g than `Vec2d` as well, IMO.

rfuest, 2022-01-06:
> `Line::polar` (or `Line::new_polar`) might be useful too

It would be useful IMO.

Or we could add a generic polar to rectangular coordinate conversion to `Point`:
```rust
let delta = Point::from_polar(45.0.deg(), 20);
let line = Line::with_delta(Point::new(1, 2), delta);
```

rfuest, 2022-01-06:
:thinking:  But that isn't easily possible because `Angle` isn't included in core.

bugadani, 2022-01-06:
What does `Point::from_polar(45.0.deg(), 20)` mean? By itself I feel it's terribly confusing.

rfuest, 2022-01-06:
> Edit: I think I might have figured it out. Disregard me.

OK, I will, but if you have a better idea I'm open for suggestions. We could, for example, add a new `Polar` struct:
```rust
struct Polar {
    angle: Angle,
    distance: u32,
}
```
and implement `From` to convert between `Point` and `Polar`. This would also fix the problem that `Angle` is not include in core.

bugadani, 2022-01-06:
It's possible to hide an extension trait in prelude and sidestep the `Angle` issue that way, if I understand the problem correctly. But after my `polar` brainfart, who knows 🤔

I think I prefer `from_polar` over a `Polar` struct. `Polar` just sounds like a special cased vector, which circles back to my other comment on this issue. So IMO the cleanest route forward would be to stick to `Point` and add an extension trait with `from_polar()`.

rfuest, 2022-01-06:
> It's possible to hide an extension trait in prelude and sidestep the `Angle` issue that way, if I understand the problem correctly.

That would be a solution. With the limitation that trait methods can't be const. But a const constructor might not be possible anyway (haven't looked into this).

> I think I prefer `from_polar` over a `Polar` struct. `Polar` just sounds like a special cased vector, which circles back to my other comment on this issue. So IMO the cleanest route forward would be to stick to `Point` and add an extension trait with `from_polar()`.

A `Polar` struct might be nicer as a return type from a rectangular to polar conversion. Otherwise `to_polar` must return `(Angle, u32)`.

Not distinguishing between points and vectors was a decision we made back when we replaced the old `Coord` and `UnsignedCoord` types. Making this distinction would be mathematically cleaner, but I cannot think of any issues we had with this discussion since it was implemented. Can you think of any parts of the API that would be cleaner/safer if we introduced the distinction between points and vectors?

If we implement `Polar` it would also be defined as a point with the given angle and distance from the origin and not a vector.

rfuest, 2022-03-25:
I've changed the title of this issue, because `Line::with_delta` was added in #641 and the discussion is now about polar coordinates.