embedded-graphics/embedded-graphics #618: rust-psp linking with rust-lldfailed

FloVanGH, 2021-08-06:
- Version of embedded-graphics in use (if applicable): 0.7.1
- Target: PSP
- If using real hardware: rust-psp

https://github.com/overdrivenpotato/rust-psp

## Description of the problem/feature request/other

I try to update the version of embedded-graphics in the rust-psp crate. But I cannot compile the example. If there is now `.draw` call in the code it compiles otherwise not. Check https://github.com/overdrivenpotato/rust-psp/issues/102

## Test case (if applicable)

* Clone https://github.com/FloVanGH/rust-psp
* Check the readme for the setup
* Navigate to `examples/embedded-graphics` and run `cargo psp`.

rfuest, 2021-08-06:
Try to enable the `fixed_point` feature and check #616 for a related discussion.

FloVanGH, 2021-08-06:
Worked thank you :-)

closed 2021-08-06 by FloVanGH