embedded-graphics/embedded-graphics #633: add Korean Font to embedded-graphics

jjpark78, 2021-12-19:
is embedded-graphics support add CJK language fonts ? 

if so, is there a tutorial or documents about it ?

jamwaffles, 2021-12-21:
I believe it could be added as long as Korean is written left-to-right - we don't currently support RTL text unfortunately. All the builtin fonts are generated from the BDF files [here](https://gitlab.freedesktop.org/xorg/font/misc-misc/-/tree/master) so if some or all of those support the Korean character set, it could be added. 

There's no tutorial, but a starting point would be the `convert-fonts` command in the Justfile which runs `tools/convert-fonts/src/main.rs`. I believe this uses the mappings defined in the `impl_mapping!` macro in `src/mono_font/mapping.rs` if you wanted to take a look at implementing it yourself. 

If the source fonts don't support Korean characters or some other issue comes up, I'd suggest making a separate crate with any required features to better support Korean

ferristseng, 2022-02-07:
I'm working on some mono fonts for Chinese, but could easily add Korean. See: https://github.com/ferristseng/rust-embedded-graphics-cjk

Feel free to open an issue there! I may need some help since I don't speak Korean :)

flyingyizi, 2022-10-29:
[embedded-fonts](https://github.com/flyingyizi/embedded-fonts) may be your wanted. This project includes a small tool convert-bdf to split(optional) and convert the BDF file format into the internal data structure for text render by the [embedded-graphics] library.  usage refer the https://github.com/flyingyizi/embedded-fonts/examples.

througth it, you can only import ness
you also can use ttf and pcf fonts. how to convert them to bdf is written in the readme. 

With this crate, you can only import the required glyphs into your project. For example, only 35 characters "中国欢迎China welcomes日本へようこそWelcome to Japan북한 환영Welcome North Korea" are introduced, which is very meaningful in MCUs with limited space, such as avr.