embedded-graphics/embedded-graphics #643: [idea] Add QOI file format

sbechet, 2022-01-12:
Hello,

It can be interesting to add QOI file format like png or bmp because code is really simple, secure and fast for embedded projects:

source:

- [blog](https://phoboslab.org/log/2021/12/qoi-specification)
- [qoiformat.org](https://qoiformat.org/)
- [reference code](https://github.com/phoboslab/qoi)

no-std crates:

- [rapid-qoi](https://crates.io/crates/rapid-qoi)
- [qoi](https://crates.io/crates/qoi)

rfuest, 2022-01-13:
Thanks for the suggestion. I hadn't heard about this format, but it seems great for embedded applications.

And it's easy to implement: https://github.com/embedded-graphics/tinyqoi. The crate still needs some polish but it works as a proof of concept.

jamwaffles, 2022-01-13:
Nice! Looks like you got nersniped ;)

rfuest, 2022-01-14:
The first version `tinyqoi` is now released: https://crates.io/crates/tinyqoi

sbechet, 2022-01-18:
Hello Ralf,

thank you very much, I hope that [tinyqoi](https://crates.io/crates/tinyqoi) will be useful for your projects, in any case, it seems nice to me!
I'm closing the issue since it no longer makes sense.
Have a good day !

closed 2022-01-18 by sbechet