embedded-graphics/embedded-graphics #646: [Question] Is there a way to clip the drawn object?

derchr, 2022-01-20:
- Version of embedded-graphics in use (if applicable): 0.7
- Target: ESP32
- If using real hardware: ssd1306

https://crates.io/crates/ssd1306

## Description of the problem/feature request/other

Hello,
I'm trying to implement a (horizontally) scrolling text. My idea was to just render the text as a long string that does not fit on the display and clip it so that it only renders at the place, where I want it to be. Then I just move the position where the text is rendered.
Is there a way to achieve this? I saw the clipped method on DrawTargetExt, but I'm unsure if this applicable for my use case as this restricts the drawable area of the whole display.

Thank you!

jamwaffles, 2022-01-20:
The `clipped` method is probably what you want. It gives you a _new_ draw target which will clip anything drawn to it. The original `display` can be used as normal. If you haven't seen it, there's an example [here](https://docs.rs/embedded-graphics/latest/embedded_graphics/draw_target/trait.DrawTargetExt.html#examples-2) to draw some static text into a clipped target. If you have and it's confusing, please let us know!

derchr, 2022-01-21:
Right, the `clipped` method is exactly what I want. I have a wonderful scrolling text now.

Somehow, yesterday I couldn't wrap my head around the fact that I can pass different `DrawTargets` to different `Drawables`.
Thanks you!

closed 2022-01-21 by derchr

jamwaffles, 2022-01-21:
No problem! Glad it's fixed :)

thomas725, 2023-07-07:
Can you share a code sample? I don't understand exactly what you did but it sounds like it's either similar or even the same thing I want to achieve:
* I have a tiny 128x32 pixel screen that I have setup so it's possible to use it with this library
* I would now like to write a single line of text (that uses the whole 32 pixels height) that is too long to display all at once
* worded differently: I'd like to shift the buffer to the left 1px every ~200ms (or maybe different timing, would need to test how that looks) and fill the new space emerging on the right with a new character as soon as the last added character did fit into the available space completely.