embedded-graphics/embedded-graphics #658: How to rotate the Text?

hanusek, 2022-03-23:
- Version of embedded-graphics in use (if applicable): 0.7.1
- Target: real hardware
- If using real hardware: OLED 1.3'' I2C
.
## Description of the problem
I would like render two text objects. The first - horizontal, the second - vertical.
How to do it?

rfuest, 2022-03-23:
Which type of vertical text do you mean?
[image] or [image]

There is no builtin support for either variant at the moment. But you can archive both by using some additional code. To implement the 2nd version you can implement a `DrawTarget` adapter, which rotates all drawing operations:
```rust
impl<'a, T> RotatedDrawTarget<'a, T>
where
    T: DrawTarget,
{
    pub fn new(parent: &'a mut T) -> Self {
        Self { parent }
    }
}

impl<T> DrawTarget for RotatedDrawTarget<'_, T>
where
    T: DrawTarget,
{
    type Color = T::Color;
    type Error = T::Error;

    fn draw_iter<I>(&mut self, pixels: I) -> Result<(), Self::Error>
    where
        I: IntoIterator<Item = Pixel<Self::Color>>,
    {
        let parent_width = self.parent.bounding_box().size.width as i32;

        self.parent.draw_iter(
            pixels
                .into_iter()
                .map(|Pixel(p, c)| Pixel(Point::new(parent_width - p.y, p.x), c)),
        )
    }
}

impl<T> Dimensions for RotatedDrawTarget<'_, T>
where
    T: DrawTarget,
{
    fn bounding_box(&self) -> Rectangle {
        let parent_bb = self.parent.bounding_box();
        Rectangle::new(
            parent_bb.top_left,
            Size::new(parent_bb.size.height, parent_bb.size.width),
        )
    }
}
```
And use it like this:
```rust
  Text::new(
        "Rotated text",
        Point::new(15, 15),
        MonoTextStyle::new(&FONT_5X8, BinaryColor::On),
    )
    .draw(&mut RotatedDrawTarget::new(&mut display))?;
```

hanusek, 2022-03-23:
> ```rust
> RotatedDrawTarget
> ```

Thanks for reply.
Look 
[image]

rfuest, 2022-03-23:
Here is an example how this could be implemented in e-g: https://gist.github.com/rfuest/461b47aa16de08aa21be5428a107b9f7

[image]

hanusek, 2022-03-23:
@rfuest You are great! Thx!

closed 2022-03-23 by hanusek