embedded-graphics/embedded-graphics #670: Add custom font

simon0356, 2022-04-04:
- Version of embedded-graphics in use (if applicable): 0.7
- Target: real hardware
- If using real hardware: Waveshare epaper 4.2"

Please link to the driver on [<https://crates.io>](https://crates.io/crates/epd-waveshare) if applicable.

## Adding custom font 
I would like to use helvetica bold font but i can't figured out how to export .raw file to import it in code as seen in example : 
https://github.com/embedded-graphics/examples/blob/main/eg-0.6/examples/text-custom-font.rs
I there any documentation to do so ? 

## Test case (if applicable)

```rust
// Failing test case demonstrating the issue here
```

Thanks
Simon

rfuest, 2022-04-05:
> I there any documentation to do so ?

No, unfortunately there isn't any documentation on creating font images for use with `MonoFont`.

I've recently resumed working on adding support for BDF fonts to embedded-graphics. This should provide a more accurate display of Helvetica, because the new text renderer supports proportional fonts. And it will also contain some tooling to make it easier to convert BDF fonts into images for `MonoFont`.

But until that is available I've created some monospaced versions of Helvetica, which are based on the xorg BDF files at https://gitlab.freedesktop.org/xorg/font/adobe-100dpi.

[image]

Download: [helvetica.zip](https://github.com/embedded-graphics/embedded-graphics/files/8419718/helvetica.zip)

simon0356, 2022-04-05:
Thanks, i successfully run your example, i will use your exported font for my project.

Have a good day

simon0356, 2022-04-20:
I'm sorry to add comment on a closed issue, but i have tried to convert myself the bdg to .raw and .png file and i can get the same .raw file md5sum, and the font did not display as it should on screen 
I use the following with the .bdg file in xorg repository linked 
cargo run helvB10.bdf helvb10 --png helvb10.png --raw helvB10.raw

Could you give the command used to generate .raw and .png file ?

Thanks
Simon

rfuest, 2022-04-20:
> I'm sorry to add comment on a closed issue,...

No problem, but I would prefer if you reopen the issue if the problem reappeared. I think you should have the permission to do so if you closed the issue yourself, otherwise ignore what I said.

I did use a new font converter, which isn't released yet. But I noticed a bug in `bdf-to-mono`, which I hadn't used in a long time. Try to convert the font using the fixed version in #672.

simon0356, 2022-04-20:
Thanks, md5 match now :) 
In fact i have tried to perform the conversion myself to generate the .raw file for helvB08.bdf  font, which is generate the following assert error : 
```
simon@simon-msi:~/rust/embedded-graphics/tools/bdf-to-mono$ RUST_BACKTRACE=1 cargo run ~/Téléchargements/helvB08.bdf helvB10 --png helvB08.png --raw helvB08.raw 
    Finished dev [unoptimized + debuginfo] target(s) in 0.04s
     Running `/home/simon/rust/embedded-graphics/tools/target/debug/bdf-to-mono '/home/simon/Téléchargements/helvB08.bdf' helvB10 --png helvB08.png --raw helvB08.raw`
thread 'main' panicked at 'assertion failed: `(left == right)`
  left: `-1`,
 right: `0`', bdf-to-mono/src/lib.rs:128:9
stack backtrace:
   0: rust_begin_unwind
   1: core::panicking::panic_fmt
   2: core::panicking::assert_failed_inner
   3: core::panicking::assert_failed
             at /home/georgik/projects/rust/library/core/src/panicking.rs:138:5
   4: bdf_to_mono::MonoFontData::new
             at ./src/lib.rs:128:9
   5: bdf_to_mono::main
             at ./src/main.rs:32:18
   6: core::ops::function::FnOnce::call_once
             at /home/georgik/projects/rust/library/core/src/ops/function.rs:227:5
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```
Is there a minimum size for a font ? 

Thanks
Simon

rfuest, 2022-04-20:
This happens because `helvB08` is a proportional fonts and some glyphs overlaps the origin to left. `bdf-to-mono` was designed for fonts with equally sized bounding boxes with no overlap to the left.

The conversion should work if you comment out the assertion in line 128 of lib.rs, but this will crop one column of pixels from some glyphs.

closed 2022-04-20 by simon0356