embedded-graphics/embedded-graphics #674: Look into more const generics stuff with Rust 1.61

jamwaffles, 2022-05-18:
Released tomorrow at time of writing, [1.61](https://github.com/rust-lang/rust/blob/stable/RELEASES.md#version-1610-2022-05-19) allows generics in `const` functions, which might allow us to `const`-ify more things if we're happy upgrading to 1.61 as our MSRV.

elpiel, 2022-06-12:
Could you elaborate on what parts of embedded-graphics do you think will benefit from const generics?

Also, do you think that there should be const generics variants for specific structs in parallel with non-const generic ones?

jamwaffles, 2022-06-12:
This issue was more of a "note to self" so apologies for the vagueness. I think my approach is/was to essentially replace `fn` with `const fn` and see how much we can leave in and still have the project compile.

elpiel, 2022-06-20:
Ahhh ok. Becuase of the `const generics` part I though you'll be looking into replacing parameters with const generics, e.g. for `Size<const W: usize, const H: usize>`.

Do you see any benefit in doing so in the foreseeable future?

jamwaffles, 2022-06-20:
We might see some performance benefits when used in internal code, but I don't think it would benefit the public API to add const parameters everywhere

Finomnis, 2022-08-28:
I don't think there is much use to `Size<const W, const H>` until you can actually do math with it. Right now you can't even create a `[0u8; W*H]` with that information.

rfuest, 2022-09-01:
Just to clarify something: I think this issue was not intended to be about const generics, but being able to use trait bounds and impl Trait return values together with const fns. These features were stabilized in version 1.61.0, which is mentioned in the original post.

closed 2022-09-02 by jamwaffles