embedded-graphics/embedded-graphics #678: Mono font rendering

elpiel, 2022-06-12:
- Version of embedded-graphics in use (if applicable): 0.7
- Target: simulator

## Description of the problem/feature request/other

I've come to realize while trying to make precise calculations with text height, that the Mono font rendering size differs from what I expected.

I couldn't open the `.raw` font files but it seems to me that even using the baseline as a character height leaves a few pixes above, when rendering them.
Is there something I'm missing or haven't viewed that can shed some light on this issue?

## Test case (if applicable)

```rust
use embedded_graphics::{
    mono_font::{ascii::FONT_10X20, MonoTextStyle},
    pixelcolor::Rgb555,
    prelude::*,
    text::Text,
};

use embedded_graphics_simulator::{OutputSettingsBuilder, SimulatorDisplay, Window};
use embedded_graphics::{ mono_font::{ascii::FONT_10X20, MonoTextStyle}}
fn main() -> Result<(), Box<dyn std::error::Error>> {
    let mut display = SimulatorDisplay::<Rgb555>::new(Size::new(320, 240));

    let font = FONT_10X20;
    let color = Rgb555::WHITE;
    let character_style = MonoTextStyle::new(&font, color);

    Text::new("1234567890", Point::new(0, font.baseline as i32), character_style)
        .draw(&mut display)
        .unwrap();
    Text::new("ABCDEFG", Point::new(130, font.baseline as i32), character_style)
        .draw(&mut display)
        .unwrap();

    let output_settings = OutputSettingsBuilder::new()
        .scale(3)
        .pixel_spacing(0)
        .build();

    Window::new("Text size check", &output_settings).show_static(&display);
}
```
[image]

rfuest, 2022-06-12:
The fonts are also included as PNG files in the repo: https://github.com/embedded-graphics/embedded-graphics/blob/master/fonts/png/ascii/font_10x20.png

You are correct that all glyphs in the ASCII subset have empty pixel rows at the top. But there are glyphs from other subsets, like `Ź` from https://github.com/embedded-graphics/embedded-graphics/blob/master/fonts/png/iso_8859_16/font_10x20.png which use the upper rows of the glyph bounding box.

elpiel, 2022-06-12:
Do you know which font sizes have how many pixels offset?
I'd like to make some precise calculations based on the font size so I have to take into account the empty rows for the ASCII subset.

rfuest, 2022-06-12:
No, I don't, this information isn't contained in the source BDF files. You will either need to count the empty pixels manually or write some code to determine the offset.

rfuest, 2022-06-12:
Sorry, my previous post wasn't correct. The BDF files contain the height of the uppercase letter: https://github.com/embedded-graphics/embedded-graphics/blob/7db5e737a366e26b37e56681bb5d3162c1e822cc/fonts/src/5x7.bdf#L29
With this information you should be able to calculate the necessary offset.

elpiel, 2022-06-12:
Ohhh. Ok, then at least the information is there! now I need to do a font that only takes this height into account.

closed 2022-06-12 by elpiel