embedded-graphics/embedded-graphics #681: Trait for double buffered screens

mchodzikiewicz, 2022-06-27:
I started working on a driver for STM32's LTDC that would utilize this peripheral a bit better than examples from `stm32f4xx_hal` and `stm32f7xx_hal`.

Since these can handle quite sizable screens, process of drawing can take long enough to make it noticeable to the user. It would be nice to have an extension trait (in `embedded-graphics-core` I guess) that allows to indicate to your `DrawTarget` that you have finished creating a frame and want to display it. (It would also be a place to hide buffer swapping and other implementation details).

It would also add a possibility to pass control over switching buffers to some more advanced widgets (like a plot that needs to draw a lot of primitives that only makes sense in a batch)

My WIP is just something like
```rust
pub trait DoubleBuffered
    where
        Self: DrawTarget,
{
    fn commit(&mut self) -> Result<(), Self::Error>;
}
```
but let me implement the driver first to make sure there is noting more needed from it.

In the meanwhile, I'd like to share it to possibly gather some more insights

rfuest, 2022-08-06:
Is it necessary to have a separate `DoubleBufferer::commit` method if a `flush` method is added? This change is discussed in #647.

Regarding the layers: IMO each layer should be exposed as a separate `DrawTarget`, which wouldn't require any changes to e-g. But it would require that at least some color types with transparency would be added to e-g-core.

mchodzikiewicz, 2022-08-07:
Yes, `flush` discussed in #647 is basically the same feature, I won't argue about what is a better name for it :)

Regarding layering (and possibly some other display capabilities), the only reason I feel it might be beneficial to have a way to express layering capabilities in `embedded-graphics-core` is that it could pass the contract between display driver and possible higher level frameworks lowering the cognitive load for a person that just wants to create some UI layout.

In other words, framework built upon `embedded-graphics` could statically infer the display capabilities and make some optimizations based on them.

I am far from proposing concrete solution here, but perhaps some higher level abstraction trait for the physical display could do. It could have a list of draw targets that user can borrow, some abstracted configuration methods and so on

rfuest, 2022-08-08:
> Yes, `flush` discussed in #647 is basically the same feature, I won't argue about what is a better name for it :)

OK, great. We'll use `flush`, because most requests for this feature used this term.

> Regarding layering (and possibly some other display capabilities), the only reason I feel it might be beneficial to have a way to express layering capabilities in `embedded-graphics-core` is that it [...]

This sounds interesting and is something we should investigate. The next major version of e-g-core needs to be designed with better support for hardware acceleration in mind. Up until now most development was done using the simulator or small SPI display. For the next version I had planned to play with features like LTDC and DMA2D before settling on a new API. This will take some effort to get right and I'm not sure if a higher level layering abstraction will be part of the first version of the new API.

mchodzikiewicz, 2022-08-08:
Sure, it is better not to provide an API than to provide an ill abstraction and break it with consequent releases. Especially if it is targeted to be used in bigger projects.

Now that I think about it, I guess it should start as an external crate and be internalized when at least few display drivers would implement it.

Side note: if there is an ongoing research on API, it might be worth to take into account drivers that draw primitives on their own, such as FT80x. It is a bit similar approach to what OpenGL does. Perhaps OpenGL display driver could also be a nice exploration target.

rfuest, 2022-08-08:
> Side note: if there is an ongoing research on API, it might be worth to take into account drivers that draw primitives on their own, such as FT80x. It is a bit similar approach to what OpenGL does. Perhaps OpenGL display driver could also be a nice exploration target.

There actually used to be support for this in the past, but we removed it in e-g 0.7. The reason for the removal was that it isn't trivial to draw primitives in the exact way or at least very close to the default implementation. We noticed that the few drivers which tried to implement a subset of the primitive drawing function all contained bugs that would cause an incorrect display for some style options. At that point we decided to simplify the `DrawTarget` API to make it easier to implement. Consistency between different drivers was more important than performance to us.

IMO an abstraction layer for FTDI/Bridgetek ICs, OpenGL or similar targets could be implemented as a separate crate, where e-g could be used as the software rendering backend. It could reuse the geometry and color types from e-g, but handle drawing differently.