embedded-graphics/embedded-graphics #683: `fixed` MSRV problems

jamwaffles, 2022-08-14:
The build in #682 failed because `fixed` requires Rust 1.61 from version [1.16.0](https://gitlab.com/tspiteri/fixed/-/blob/master/RELEASES.md#version-1160-2022-06-30). We could:

1. Bump our MSRV again to whatever the latest version of Rust is at the time the bump is done.
2. Change our `Cargo.toml` to require 1.15 or less.

I think I'd prefer 1. as development on e-g has slowed a lot recently due to my focusses being elsewhere.

Finomnis, 2022-08-28:
I agree with #1. Ideally, none of this would be necessary, because a change in MSRV would be a breaking change in SemVer. But I guess that's out of our control for dependency crates.

jamwaffles, 2022-09-01:
Number 1 would also open the door to #674 so two birds, one stone...

rfuest, 2022-09-01:
SGTM

PR #689 (2022-09-01): Bump MSRV to 1.61
Thank you for helping out with embedded-graphics development! Please:

- [x] Check that you've added passing tests and documentation
- [x] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, etc) if your changes affect the **public API**
- [x] Run `rustfmt` on the project
- [x] Run `just build` (Linux/macOS only) and make sure it passes. If you use Windows, check that CI passes once you've opened the PR.

## PR description

Closes #683 

For a speed boost, I also changed the CircleCI resource class to "large". I can't  [... cut]
closed 2022-09-01 by jamwaffles