embedded-graphics/embedded-graphics #686: Improve font renderer capabilities

Finomnis, 2022-08-30:
With [re-implementing the U8g2 font subsystem](https://github.com/Finomnis/u8g2-fonts) I now have an idea of what the capabilities of an established font rendering algorithm are.

I published it as a separate crate and with no integration to [`embedded_graphics::Text`](https://docs.rs/embedded-graphics/0.7.1/embedded_graphics/text/struct.Text.html). The reason is that there are several shortcomings of the font rendering interface of `embedded_graphics` that I'd like to add in the future:
 - [ ] Compatibility with `format_args!()` to avoid the necessity for temporal string allocations when rendering arguments
 - [ ] Support for newlines
 - [ ] Return rendered dimensions from text drawing functions

Sadly, the first point already (`format_args!()`) is completely incompatible with the current API, because the current API is based o the builder pattern, and `format_args!()` does not live long enough for chained calls.

I'd like to improve the existing API, so I hope to have a little bit of a discussion here. (@rfuest)

If we provide an API similar to [`u8g2_fonts::FontRenderer`](https://docs.rs/u8g2-fonts/latest/u8g2_fonts/struct.FontRenderer.html), then I think the actual abstract Renderer trait could be reduced to a `renderGlyph` and several `getAscent`, `getDescent`, etc.

Finomnis, 2022-08-30:
In general you are right that a `TextWriter` struct could be used, but that is only possible for left aligned strings. For centered or right-aligned you need to know the size of the entire string beforehand.

So what I read from your comments is that you would rather keep it as-is and declare that `utf8-fonts` is an external font renderer explicitly?

Finomnis, 2022-08-30:
Removed the newlines comment

rfuest, 2022-08-30:
> In general you are right that a `TextWriter` struct could be used, but that is only possible for left aligned strings. For centered or right-aligned you need to know the size of the entire string beforehand.

OK, centered text would be a problem. But I still think that it should be possible to implement it similar to your implementation in `u8g2-fonts`.

> So what I read from your comments is that you would rather keep it as-is and declare that `utf8-fonts` is an external font renderer explicitly?

I assume you mean `u8g2-fonts`?

I apologize if my comments made you think that I just want to dismiss your suggestions. Finding a common solution that works for all text renderers would be the best solution. I prefer to keep the required API changes to a minimum, but if you convince me that bigger API changes are required we could implement them for e-g 0.8.

Finomnis, 2022-08-30:
Sounds good :) I'd be course be willed to adjust my api, with the premise that I don't want to lose any features in the process.

Finomnis, 2022-08-30:
As a little background: the biggest issue was not just the alignment, it was the multi-line alignment without allowing memory for caching the line widths. Because you need to render each line twice, once for width, and once for the actual aligned rendering. And you can't easily re-render the current line with `format_args!()`

rfuest, 2022-08-30:
> As a little background: the biggest issue was not just the alignment, it was the multi-line alignment without allowing memory for caching the line widths. Because you need to render each line twice, [...]

Hmm, that's a problem. I did take a quick look at the implementation in `u8g2-fonts`. Am I correct that `ArgsLineDimensionsIterator` is used to implement the caching of the line width? That struct is pretty large and will, in many cases, consume more memory than a string buffer.

Perhaps we could limit `format_args` to single line rendering, or at least offer single and multi line variants. I suspect that single line rendering would be the most common use case and it should be as memory efficient as possible. A multi line variant could also use const generics to set the maximum number of lines to keep the line dimensions buffer size smaller.

Finomnis, 2022-08-30:
The line dimensions buffer is 20 right now. I felt like this is a good compromise. If there are more than 20 lines, the input string is read again. So it's not actually that big. It can further be optimized to be `20*i32` instead of `20*RenderedDimensions`. So I personally thought that is quite small compared to the amount of memory that gets moved to do the actual rendering.

If you have a const generic, then you basically move the buffer allocation back to the user. And then the use could as well allocate a string buffer and `write!()` to it directly and render with a `&str`. That's at least my thought.

Are you sure that all the things you talk about here are actually performance relevant? Because my intuition says the slowest part is the glyph lookup and the glyph rendering. And if you have the kilobytes of memory that a font requires, I don't think 80 bytes of line buffer really matter, or the time to fill that buffer.

So before basing an entire architecture on the premise that this is not fast enough, I would honestly benchmark.

rfuest, 2022-08-31:
My comment was related to memory consumption and not performance. Although benchmarking the performance of having to format the string multiple times is a good idea.

The font data is constant and can be stored in flash on an embedded system and don't require any RAM. But the line dimensions will end up on the stack and `20*RenderedDimensions` is big for a microcontroller with a few kilobytes of RAM.

Finomnis, 2022-09-01:
@rfuest I reduced the cache down to 5*(i32,u32,i32), meaning, 60 bytes. Do you think this is acceptable for most microcontrollers?

https://github.com/Finomnis/u8g2-fonts/pull/3

rfuest, 2022-09-01:
Yes, I think 60 bytes are fine. Users with even more memory constrained targets should probably not be using `core::fmt` anyway and switch to something more memory efficient like `ufmt`.

Finomnis, 2022-09-07:
@rfuest Just released v0.2.0 of u8g2-fonts, where I integrate the embedded_graphics::text subsystem. So it's no longer just an external renderer.

I'd close this discussion section for now, if that's fine with you.

closed 2022-09-07 by Finomnis