embedded-graphics/embedded-graphics #687: `MonoFont: Send + Sync`

kpreid, 2022-08-31:
- Version of embedded-graphics in use (if applicable): 0.7.1

## Description of the feature request

[The `MonoFont` struct](https://docs.rs/embedded-graphics/0.7.1/embedded_graphics/mono_font/struct.MonoFont.html) contains a field

```rust
    pub glyph_mapping: &'a dyn GlyphMapping,
```

which causes the whole `MonoFont` to be `!Send + !Sync`, meaning that it cannot be stored or referenced in a `static` variable, or sent between threads. Is this an intended restriction? If not, I'd like to request that it be made `Send + Sync`, either by changing the field type to 

```rust
    pub glyph_mapping: &'a dyn GlyphMapping + Send + Sync,
```

or by changing the trait definition,

```rust
pub trait GlyphMapping: Send + Sync {
```

whichever seems more appropriate. This should not be onerous since the mapping should generally be a stateless function. (But it is a semver breaking change that could affect any code that constructs a `MonoFont` or implements `GlyphMapping`.)

I would be happy to provide a PR to change this, but there's at least one design choice consideration, so I'm presenting it as an issue first.

## Test case

```rust
use embedded_graphics::mono_font::{
    ascii::{FONT_9X15, FONT_9X15_BOLD},
    MonoFont,
};

static FONT_LIST: [MonoFont<'_>; 2] = [FONT_9X15, FONT_9X15_BOLD];
```

```
error[E0277]: `(dyn GlyphMapping + 'static)` cannot be shared between threads safely
 --> src/lib.rs:6:19
  |
6 | static FONT_LIST: [MonoFont<'_>; 2] = [FONT_9X15, FONT_9X15_BOLD];
  |                   ^^^^^^^^^^^^^^^^^ `(dyn GlyphMapping + 'static)` cannot be shared between threads safely
  |
  = help: within `[MonoFont<'static>; 2]`, the trait `Sync` is not implemented for `(dyn GlyphMapping + 'static)`
  = note: required because it appears within the type `&'static (dyn GlyphMapping + 'static)`
  = note: required because it appears within the type `MonoFont<'static>`
  = note: required because it appears within the type `[MonoFont<'static>; 2]`
  = note: shared static variables must have a type that implements `Sync`
```

rfuest, 2022-08-31:
Sounds like a reasonable change. I think I would prefer to change the trait definition.

Just out of curiosity: Why do you want to store fonts in a static variable in the first place, instead of using a const?

kpreid, 2022-09-01:
> Why do you want to store fonts in a static variable in the first place, instead of using a const?

I can't say I do; it was just the shortest test-case I could think of to demonstrate the behavior without pulling in `std`.

My actual use case is building up a UI widget-tree structure in a library that wants to be thread-_compatible_, so I have `Arc<dyn Widget>` and `trait Widget: Send + Sync`; I wanted a text widget that can be given a choice of font (`&'static MonoFont`), and then discovered that this failed to compile.

(Arguably, `embedded-graphics` isn't the library I should be using; I picked it originally based on the requirements "low-resolution pixel-by-pixel drawing to arbitrary devices" and "draws text using _included_ bitmap fonts" — but not required no_std or low resource usage. But I figure I might as well continue, until such time as I find an actual conflict between my goals and `embedded-graphics`'s. Hence my tentative approach to the original request.)

PR #688 (2022-09-01): Require `GlyphMapping` implementors to also be `Sync`.
- [X] Check that you've added passing tests and documentation
- [x] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, etc) if your changes affect the **public API**
- [X] Run `rustfmt` on the project
- [X] Run `just build`

## PR description

Added `Sync` as a supertrait to `GlyphMapping`. This means that the `&dyn GlyphMapping` in `MonoFont` no longer causes `MonoFont` to be `!Send + !Sync`. Now, it is `Send + Sync`.

This is a semver breaking change because it imposes a new requ [... cut]
closed 2022-09-01 by jamwaffles