embedded-graphics/embedded-graphics #692: Font Renderer: vertical alignment for multi-line strings

Finomnis, 2022-09-04:
I noticed that multi-line strings always get aligned at the first line. This is fine for `Baseline::Top` and `Baseline::Alphabetic`, but I would have expected it to be aligned for the entire string for `Baseline::Bottom` and `Baseline::Middle`.

Implementing this should be fairly simple; iterating through the input string once to count the number of newlines should be very minor compared to the amount of computation required to do the actual rendering.

Is this an active design decision or was this just never implemented?
@rfuest

jamwaffles, 2022-09-18:
I believe the current `Baseline` behaviour is correct; it should only shift the origin relative to the first line, e.g. `Baseline::Bottom` uses the bottom of the first character's bounding box as the text origin:

[image]

If you want more involved alignment, I'd recommend [embedded-text](https://crates.io/crates/embedded-text). It can do much more clever stuff than vanilla e-g can:

```rust
let character_style = MonoTextStyle::new(&FONT_6X10, BinaryColor::On);

let textbox_style = TextBoxStyleBuilder::new()
    .alignment(HorizontalAlignment::Left)
    .vertical_alignment(VerticalAlignment::Bottom)
    .build();

TextBox::with_textbox_style(
    "This is a\nmultiline\nHello World!",
    display.bounding_box(),
    character_style,
    textbox_style,
)
.draw(&mut display)?;
```

rfuest, 2022-09-19:
> Is this an active design decision or was this just never implemented?
> @rfuest

As @jamwaffles already answered: This behavior is intentional.

Finomnis, 2022-09-19:
Got it.

closed 2022-09-19 by Finomnis