embedded-graphics/embedded-graphics #696: Should `ImagePixelGetter` be renamed?

Nerixyz, 2022-10-01:
As discussed in embedded-graphics/tinybmp#35:

I think `ImagePixelGetter` (added in #612) should be named something like `GetPixel` which is more Rust-like (when comparing to traits from std: `FromStr`, `AddAssign`, `IntoFuture`).

rfuest, 2022-10-11:
@jamwaffles I would like to hear your opinion on this. Do we have any other traits with similar naming conventions?

jamwaffles, 2022-10-11:
I wouldn't be opposed to a rename. A quick peruse shows we already have stuff like `Drawable`, `DrawTarget`, `AngleUnit`, `OriginDimensions`, `WebColors`, etc. I can't think of a better name than `GetPixel`, but I feel there should be something in it that relates to the source data being a 2D array of pixels, as opposed to e.g. a primitive that generates its pixels dynamically. `GetImagePixel`? Still sounds weird...

rfuest, 2022-10-12:
I think that `GetPixel` would be my favorite name if the trait wouldn't depend on the super trait `ImageDrawable`. We could change the trait to include an associated color type instead of reusing `ImageDrawable::Color`:
```rust
pub trait GetPixel {
    type Color: PixelColor;
    fn pixel(&self, p: Point) -> Option<Self::Color>;
}
```
This would make the trait more flexible and we could also use it for `Framebuffer::pixel` (`Framebuffer` doesn't implement `ImageDrawable`)

jamwaffles, 2022-10-14:
Sounds good to me!

closed 2022-10-16 by rfuest