embedded-graphics/embedded-graphics #700: Move `FrameBuffer` into core

jamwaffles, 2022-10-17:
I'm not sure if this can be done given the imports it depends on, but it occurred to me that display drivers which depend on e-g-core would want to use `FrameBuffer` without depending on e-g.

jamwaffles, 2022-10-19:
> Do you think that Framebuffer would be exposed in the API of a display driver?

No, just used internally. It feels odd to require driver authors to depend on e-g _and_ e-g-core but maybe I'm thinking about this the wrong way.

rfuest, 2022-10-19:
They don't need to depend on e-g-core, if they don't want to. The driver would still remain compatible, because e-g reexports the types from e-g-core.

But by depending on both crates and only using e-g-core in the display driver API it is easier to ensure that no incompatible types are introduced into the API.

jamwaffles, 2022-10-19:
Thanks for explaining. I gave this a bit more thought and you're right, it's not really an issue for drivers. I'll close _this_ issue :)

closed 2022-10-19 by jamwaffles

rfuest, 2022-10-19:
OK. I just also realized that another reason to not move `Framebuffer` to e-g-core is that it depends on `ImageRaw`, which isn't included in e-g-core.

jamwaffles, 2022-10-19:
Good to know there's a concrete reason too :). I did look at the imports it uses a while ago but didn't bother seeing where they came from.